supermerill/SuperSlicer #3917: unwanted 'Top Solid Infill' halfway on the front deck slope lines part of a benchy

Blinker73, 2023-08-13:
Hello, 

**Description:**
I am trying to find the cause for unwanted 'Top Solid Infill' on the front deck part of a benchy.
On some layers only, SS mixes "Top Solid Infill" and "Internal perimeter" lines halfway trough the width of the deck
Having this change in the middle of a line leaves a visible change on the final print

**Expected behavour:**
I would expect each visible part of the front deck layers to be sliced as a single line type (either one) throughout the width of the deck as Cura slicer does
[image]

**Actual behavour:**
Most of the lines SS generates for this part of the model are Internal or external perimeters
But, SS has some layers that print part of a deck line as "Top Solid Infill" mixed with "Internal perimeter" lines halfway trough the width of the deck
[image]
[image]

One thing I noticed is that the layers showing the problem change if I rotate the model around the Z axis
Sliced as is, I see issues at layers 77, 79 and 87
Rotated 90deg counter clockwise, i see issues at layer 79 and 94
Rotated 180deg counter clockwise, i see issues at layer 77 and 79
Rotated 270deg counter clockwise, i see issues at layer 75, 79 and 94

What could be the cause of these inconsistent choice of line types

Thanks, 

**Desktop (please complete the following information):**
 - OS: Gentoo Linux
 - Version 2.5.59.0
 - Voron 2.4

**Additional context**
Cura version for the compare screenshot is 4.13.1

[SaveAs.zip](https://github.com/supermerill/SuperSlicer/files/12330401/SaveAs.zip)

legend069, 2023-10-21:
this happens from your unusual extrusion width numbers.
[image]

[image]

Blinker73, 2023-10-23:
Thanks, 
I would have never found the cause.

Now I have to find out where i initially got those values because I don't remember changing them

closed 2023-10-23 by Blinker73