supermerill/SuperSlicer #3918: SuperSlicer2.4 bed PNG loading black

Jax1er, 2023-08-13:
### Version
_Version of SuperSlicer used goes here (help->about)_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
  * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_

supermerill, 2023-11-27:
svg works but png seems broken

supermerill, 2023-11-27:
prusaslicer 2.5 has the same problem and it's fixed in 2.7
i guess an opengl somthign taht has to be updated with newer graphics drivers.
Should be fixed after the merge.

closed 2024-02-11 by supermerill