supermerill/SuperSlicer #3920: Download, Crash on Opening macos

MexicoSkies, 2023-08-20:
### What happened?

Downloaded the non-arm version but, it crashed on opening. 

Process:               SuperSlicer [60482]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           SuperSlicer [60482]
User ID:               501

Date/Time:             2023-08-20 06:00:44.012 -0600
OS Version:            macOS 11.7.8 (20G1351)
Report Version:        12
Anonymous UUID:        EE23BCF9-E097-0A7D-1664-E4130EA5BEC9

Sleep/Wake UUID:       72F176C1-9437-4AC1-B936-05E91A3EE0D4

Time Awake Since Boot: 690000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    DYLD, [0x1] Library missing

Application Specific Information:
dyld: launch, loading dependent libraries

Dyld Error Message:
  dyld: Using shared cache: 78AC657A-5BCB-3A96-9289-6C5AE8962561
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
  Referenced from: /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
  Reason: image not found

Binary Images:
       0x109a6f000 -        0x10bb42fff +com.merill.slic3r/ (SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 - SuperSlicer_2.4.58.5_2022-09-23) <EA3F7209-F390-3268-9C90-D8081DCEDC30> /var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
       0x10dbe2000 -        0x10dc7dfff  dyld (852.2) <4FF2ABFD-E712-356A-9458-F9914B67D65C> /usr/lib/dyld
    0x7fff20544000 -     0x7fff20599fff  libc++.1.dylib (905.6) <5BA6B5ED-7842-3B13-86B0-00EB511CE2FE> /usr/lib/libc++.1.dylib
    0x7fff2065e000 -     0x7fff20afbfff  com.apple.CoreFoundation (6.9 - 1778.105) <B4B8042A-9415-3F26-91AC-735C968B0D95> /System/Library/Frameworks/CoreFoundation.framework/Versions/A/CoreFoundation
    0x7fff22d9f000 -     0x7fff22e4dfff  com.apple.framework.IOKit (2.0.2 - 1845.120.6) <A395F442-1253-3CA9-953F-7A235EEB7F67> /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit
    0x7fff2a4ee000 -     0x7fff2a4fafff  libbz2.1.0.dylib (44) <6E82D414-3810-36CF-94FF-B1BDF48DB501> /usr/lib/libbz2.1.0.dylib
    0x7fff2a6fc000 -     0x7fff2a70efff  libz.1.dylib (76.140.1) <A2FF8D14-1632-3047-9829-CC90239F97FF> /usr/lib/libz.1.dylib
    0x7fff2a7c4000 -     0x7fff2a7c5fff  libSystem.B.dylib (1292.120.1) <96E71ED1-0206-3EB1-A67C-CE44C0D0725E> /usr/lib/libSystem.B.dylib
    0x7fff2a803000 -     0x7fff2a8f3fff  libiconv.2.dylib (59) <DEE0153A-BDF9-33CA-B8C7-3C39DB906B5E> /usr/lib/libiconv.2.dylib
    0x7fff2ae4f000 -     0x7fff2ae67fff  liblzma.5.dylib (16) <455C9083-014D-3037-AC54-1395F3796734> /usr/lib/liblzma.5.dylib
    0x7fff6baf1000 -     0x7fff6bb00fff  com.apple.opengl (18.5.9 - 18.5.9) <610E765C-8C0D-3422-AD6E-636D3EBD2AD0> /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL

Model: iMac15,1, BootROM 235.0.0.0.0, 4 processors, Quad-Core Intel Core i7, 4 GHz, 24 GB, SMC 2.23f11
Graphics: kHW_AMDRadeonR9M295XItem, AMD Radeon R9 M295X, spdisplays_pcie_device, 4 GB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736453220
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1600 MHz, 0x802C, 0x384B54463531323634485A2D314736453220
Memory Module: BANK 0/DIMM1, 8 GB, DDR3, 1600 MHz, 0x85F7, 0x483634314755363747313630304700000000
Memory Module: BANK 1/DIMM1, 8 GB, DDR3, 1600 MHz, 0x85F7, 0x483634314755363747313630304700000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x142), Broadcom BCM43xx 1.0 (7.77.111.1 AirPortDriverBrcmNIC-1680.11)
Bluetooth: Version 8.0.5d7, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: APPLE SSD SD0128F, 121.33 GB
Serial ATA Device: Mercury Electra 6G SSD, 1 TB
USB Device: USB 3.0 Bus
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: FaceTime HD Camera (Built-in)
USB Device: PNY PRO ELITE PSSD
USB Device: USB Optical Mouse
USB Device: USB Receiver
Thunderbolt Bus: iMac, Apple Inc., 26.1

### Project file & How to reproduce

File appears to install properly but when opening file, immediately crashes.

### Version

2.4.58.5

### Operating system

macOS Big Sur

### Printer model

Voron 2.4

Creo2005, 2023-08-21:
Download latest night buil for macos (in Action tab) 2.5.59.2, 58.5 very old.

iAmVaeVictus, 2023-10-06:
Getting same issue 

Process:               SuperSlicer [24878]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-10-06 11:22:45.0415 -0500
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Anonymous UUID:        558C1CD0-014C-80E4-43E4-2A9A93F87FE5

Sleep/Wake UUID:       AAB1D897-256C-4643-A43F-0C7144E008D2

Time Awake Since Boot: 260000 seconds
Time Since Wake:       5590 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: <EA3F7209-F390-3268-9C90-D8081DCEDC30> /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

**Project file & How to reproduce**
File appears to install properly but when opening file, immediately crashes.

**Version**
2.4.58.5

**Operating system**
macOS Ventura

**Printer model**
Voron 2.4

MexicoSkies, 2023-10-18:
Got it! Thanks so much. Sorry for the late reply. I do have one quick question. Why is Retraction Settings in the Printer Settings and not in Printer Settings? I have to adjust for each filament. Thanks again. Love the slicer!Gary M. BrennanSent from my iPhone

MexicoSkies, 2023-10-18:
Well, I was just informed that it IS in Filament Settings. Incredible how you can look for something for hours and it’s right in front of your face. SighThanks!Gary M. BrennanSent from my iPhone

closed 2023-11-21 by supermerill