supermerill/SuperSlicer #3923: Sequential printing extruder clearance ignoring brim width

bennierex, 2023-08-26:
### What happened?

When printing squential objects, SS only takes into account the extruder clearance relative to the object itself. When using a brim, the extruder will collide when printing the brim of the next object.

### Project file & How to reproduce

[z_joints_lower.3mf.zip](https://github.com/supermerill/SuperSlicer/files/12444958/z_joints_lower.3mf.zip)

When printing the brim of instance 2, my BLtouch collides with the printed instance 1.

### Version

2.4.58.5

### Operating system

macOS Ventura 13.4.1 (22F82)

### Printer model

Ender 5 plus running Klipper

blownupp, 2023-11-21:
I’ve experienced this as well, so it’s not isolated to you

supermerill, 2023-11-22:
Brims are printed first, before any objects, and so nothing should collide.
Please help me to understand where is my mistake.

bennierex, 2023-11-22:
My brims are printed per-object...

supermerill, 2023-11-23:
oh, yes.

supermerill, 2023-11-23:
mmm.
if there is one brim per object, one skirt per object, and the skirt distance if computed from the brim, then it can be wrong.
I'll have to rework this a lot more to have something reliable. Also, i have to create a proper test case.
i'll do that after the 2.7 merge

bennierex, 2023-11-23:
@supermerill Thanks for looking into this!