supermerill/SuperSlicer #3924: app crashes when turning on spiral vase and clicking Yes

talrhv, 2023-08-28:
### Version
2.5.59

### Operating system type + version
_windows 11

### Behavior
when clicking Spiral Vase and then clicking "Yes".
the app crashes. Tried several time and always getting the same result.

_Is this a new feature request?_
No.

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[tal.zip](https://github.com/supermerill/SuperSlicer/files/12454522/tal.zip)

fprumbau, 2023-08-30:
I can confirm the issue, tried several times, Linux Fedora 38 with version 2.5.59

jaydenrussell, 2023-09-04:
Version
2.5.59.2

Operating system type + version
_windows 11

Behavior
when clicking Spiral Vase and then clicking "Yes" or loading a preset with Spiral Vase the app crashes. Tried several times and always getting the same result.

Is this a new feature request?
No.

Shampoo1014, 2023-09-11:
I know this is going to sound strange, but I had this issue some time ago and a simple way I fixed it was I either moved or rotated the file slightly. For some reason, printing that particular model in the middle of the bed had an issue. 
I haven't been able to replicate it since, and it could be a deeper issue, but you never know, it might just help :)

MechManDan, 2023-11-08:
I can confirm this issue... it is interfering withy slicer calibrations for new filaments.

supermerill, 2023-11-21:
can reproduce on 2.5.59.2
can't reproduce on nightly
I guess it's fixed for next release.