supermerill/SuperSlicer #3928: Automatic fan speed - "Enable fan if layer print is below" - should be smarter with support

CuredPrusa, 2023-09-06:
**Is your feature request related to a problem? Please describe.**
It's not a problem per se, it's more like a better way of handling printing time

**Describe the solution you'd like**
"Enable fan if layer print is below" should be smarter when using supports. Now it's actually difficult to use this function with support.

**Describe how it would work**
When the printing nozzle moves away from the actual model, to print the support, that time should be counted in the time of printing layer as well. The model surface is standing still, cooling down naturally while waiting for nozzle which is busy printing the support. What we have now is even more (too much) cooling, when nozzle returns to continue printing the actual model. 

[image]

legend069, 2023-09-20:
can you please try my repo and see if this fixes that problem?
https://github.com/legend069/SuperSlicer

supermerill, 2023-11-22:
when computing the time, it add the time of the previous support/object to the current object/support if available.

So i don't understand what's to add.

CuredPrusa, 2024-01-03:
Look at the screenshot in the first post. You see how fan goes fast in low layers because IT IS short printing time of object layer. Because this is a narrow tip of the triangle. But the whole layer there is actually quite time consuming, because the same layer has a lot of support to print. So, there is simply TOO MUCH cooling in the lower layers because the object perimeters get cool anyway, while waiting for the support to be printed.

CuredPrusa, 2024-01-03:
> can you please try my repo and see if this fixes that problem? https://github.com/legend069/SuperSlicer

Thanks. But you have no idea how lame I am in understanding github structure. I'm trying to translate it to Facebook simplicity, so I get stuck whenever I have to go beyond simple communication.  And it's often. I just have to find time and learn about it properly.

closed 2024-06-13 by CuredPrusa