supermerill/SuperSlicer #3929: Dual extruder infill

davidewen, 2023-09-07:
### What happened?

Extruder 1 will not create infill over Extruder 2 on 1st 4 layers of print like PrusaSlicer does.
This causes layer 5 to be weak and able to peel off.
Text is 0.4mm thick. 
I have 4 bottom layers

4 
[image]
[image]

### Project file & How to reproduce

[Center Guide - Crowner v6 lower body left.3mf.zip](https://github.com/supermerill/SuperSlicer/files/12550664/Center.Guide.-.Crowner.v6.lower.body.left.3mf.zip)

### Version

2.4.58.5

### Operating system

macOS Ventura 113.4.1

### Printer model

VisionMiner 22 IDEX

davidewen, 2023-09-07:
Update. 
This only happens when splitting a file into parts (which I like to do because it keeps everything together if you move them on the bed).

supermerill, 2023-11-27:
PrusaSlicer 2.5 has also kinda the same problem.
If newer version has solve it, It should be solved after merging it.
Can you reopen it if it's still here with the next 2.7 version?