supermerill/SuperSlicer #3930: Small (intentional) gaps are being widened

cy1der, 2023-09-11:
**Bug description**
SS is making gaps (that were made on purpose) wider, which is unwanted for some models (e.g I am printing a protractor, and these gaps are kinda crucial?).

**Expected behavior**
The gaps stay the same.

!!>> Project File <<!!
[Protractor.3mf.zip](https://github.com/supermerill/SuperSlicer/files/12577326/Protractor.3mf.zip)

**To Reproduce**
Steps to reproduce the behavior:
1. Slice

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Void Linux
 - Version: 2.5.59.2
 - Printer model: Irrelevant

Vanilla009, 2023-11-22:
Do you have an X-Y horizontal offset set? If so, that might be the cause of this issue.

cy1der, 2023-11-22:
Where would I find that setting, is it this?
[image]

supermerill, 2023-11-22:
XY compensation.
it shrink your entire object.

closed 2023-11-23 by cy1der