supermerill/SuperSlicer #3934: Variable layer height: Exact top surfaces

JustOneMorePrint, 2023-09-13:
**Is your feature request related to a problem? Please describe.**
Sometimes I use variable layer height to print objects exactly the height I need.
For example: Default layer height = 0.2mm, object height = 3.5mm:
I use variable layer height in the top section (with min layer height = 0.1mm) to get precise prints.
It works but could be done better.

**Describe the solution you'd like**
Please add a button like "adaptive" in the variable layer height feature to calculate a different layer height for top layers.
There should be a slider how many layers are affected and a checkbox if you want higher or lower layer heights than default.
In our example with default layer height = 0.2mm, object height = 3.5mm, now another top surface at z = 2.3mm, top layer and 3 layers below top surface will be changed, and layer height <0.2mm selected:
When clicking "Precise Top Layers" some top surfaces (and the layers beneath) will be changed in height:
4 layers with each 0.175 at 2.3mm and below will be printed. everything else at 0.2mm layer height.
(see next chapter 4 layers with each 0.175 at 3.5mm and below)

**Describe how it would work**
There should already be a kind of algorithm to filter top surfaces (because you can select infill pattern for these settings).
This could be used to determine the "problematic" surfaces. 
Multiple top layers could affect each other regarding this feature if we print the following layer at default layer height and not another (few) variable layers like in the manual method. This might take some deeper changes.

Calculation of the layer height:
selected layer count = layers to change //or available layers at default layer height between top layer and next bridge or first layer or layer with already changed variable layer height
first selected z = z top surface - (selected layer count * default layer height) 
difference = actiual z top surface in model - first selected z //here somehow the actual layer height must be taken into consideration but there could be a work around (see alternative below)
if below 0.2 is desired -> changed layer height = default layer height - (difference / selected layer count)
if above 0.2 is desired -> changed layer height = default layer height + (difference / selected layer count)

**Describe alternatives you've considered**
You might consider different layer heights for the areas below such a top surface and the area where the object will print higher.
Of course the lower part of a "virtual layer" should be printed first to avoid colision of the nozzle but this should improve the strenght of a part in that area just like when you combine infill every x layer.

**Additional context**
The "alternative idea" might also be considered as another feature to strengthen parts at perimeters and speed up quality prints a little:
Use the calculation method above to determine possible perimeter heights.
Start with small perimenter height at the outer perimeter and go up every inner perimeter until you reach default layer height.

supermerill, 2023-11-27:
That was one of the first modification i made to prusaslicer, when it was still 'Slic3r PE'. Sadly, it was ultra buggy, so i removed it. 

The first problem is to find that "exact top height", if it exists. For that, we have to identify the "top" areas (areas where there isn't a thing the next layer. It's easy. Then we have to find if there is some "flat" triangle from the stl between the top layer and the next layer that are over this top area. After getting all of these, the program may be able to find sub-areas where there is a flat reference. We can have then our 'perfect top height areas'
. when these are identify, we still have to modify the different layers to fit these. by growing some (all <= max layer height) or shrinking some ( still > min layerheight)
What if there are 2 adjacent little top areas 0.02 mm Z away and the min layer height is 0.05? Should we merge them in the bigger one or try to make weird sub-layers?

YaphetS1, 2024-10-15:
@supermerill I think we need to limit the minimum height in this case.