supermerill/SuperSlicer #3935: Errant internal perimeters going where they shouldn't.

gerthalberg, 2023-09-13:
### What happened?

I'm having trouble with errant internal perimeters going where they shouldn't. I'm using 2.5.59.2 on Windows 11 Anyone got a fix?
[image]

### Project file & How to reproduce

[Mantis Fanduct Mount L.zip](https://github.com/supermerill/SuperSlicer/files/12600431/Mantis.Fanduct.Mount.L.zip)

### Version

2.5.59.2

### Operating system

windows 11

### Printer model

Custom - corexy - Klipper

macchinette, 2023-09-17:
Disable arc_fitting

Saturn49, 2023-09-18:
I'm having the same problem.  Enabling/disabling arc fitting had no effect.

gerthalberg, 2023-09-18:
Thank's - It solved the problem for me. However I didn't disable G2/G3 arc fitting but instead I lowered the tolerance from 5 to 0.01%

closed 2023-09-18 by gerthalberg

macchinette, 2023-09-18:
> I'm having the same problem. Enabling/disabling arc fitting had no effect.

I have the last night.