supermerill/SuperSlicer #3936: Printer Cooling Modifier

Kaelum, 2023-09-18:
Different printers have different cooling abilities, and the cooling settings are primarily set in the Filament Settings. It would be nice to have the ability to specify a cooling modifier in the Printer Settings that would be multiplied by the actual cooling value to allow for normalization across printers.

**Describe the solution you'd like**
Add a setting to the Printer Settings for "Cooling multiplier".

**Describe how it would work**
A value of 1.00 would mean no change. A value of 0.60 or 60% would mean a reduction of 40% in the calculated cooling setting, or 60% of the calculated value. A value of 1.25 or "125%" would mean an increase of 25% in the calculated value.

**Describe alternatives you've considered**
Creating duplicate Filament Settings for each printer would be impossible to manage, and is currently the only possible way to manage this. My Rat Rig cooling fan is at least 50% stronger than any of my other printers. My Snapmaker is capable of less than 60% of the cooling that my K1 Max is capable of, and my Ender 5 Plus is somewhere in the middle. As you can see, this is impossible to manage.

legend069, 2023-09-20:
i can see this feature being usefull for users that have multiple printers and 1 filament profile
but i would be more inclined to add it to the "print settings" area. since printing faster requires more cooling.

supermerill, 2023-11-27:
For the whole printer of for each extruder?

Kaelum, 2023-11-27:
@supermerill I have been thinking about for quite a while, and originally thought that this would be a good idea, as it was significantly easier to maintain than having a different `Filament Setting` for each printer.  However, due to issues like those in #2895 and others, I see that what truly needs to be fixed is the naming of settings, which is related to their storage locations, which means that each printer/group needs to have a unique name for the same `Filament Setting` or `Print setting`.  This will significantly affect the coding, and is probably to big of a change for you to make, as this change needs to be a part of the core, PrusaSlicer.

Bambu Slicer and OrcaSlicer are even worse, as they completely removed the code to allow you to create  Settings files that are not related to System Settings.  So, as of now, I don't believe that the feature requested is something that would be overall helpful.

Instead, the following changes need to be made at the PrusaSlicer level:

- There should be no distinction between System and User settings.  They should be 1 and the same, with the ability to "protect" a setting from changes (which would equate to the current System settings).
- The UI needs to greatly expand to have a parent-child relationship in all settings files.  This is currently limited to allow User settings be stand-alone, or a child of a System settings file.

I'll consider submitting this to the PrusaSlicer team, but due to the amount of work that would be required, I don't believe that it is change that they will be interested in as long as PrusaSlicer is a free product.  Anywho, thank you for the consideration.

Kaelum, 2023-11-27:
#1874 sounds similar to what I'm saying.  Brainstorming again, there should probably be a core set of `Filament Settings` and `Print Settings` which are not tied to any printer.  They are core independent lists.  Each printer defined would have its own folder, which would contain its settings file along with `Filament Setting` and `Print Setting` override files for that printer.  These override files would only contain a reference to the core setting file and the settings that are being overridden.  No overrides, no settings.  The UI for this would need to be like the `Overrides` tab that currently exists under `Filament Settings`.

This is sort of what Creality did in Creality Print, but they have a ton of bugs in their system, and they mixed everything into a single setting, as that's pretty much how Cura works.  However, this would be a HUGE divergence from PrusaSlicer, thus the reason that I think the change needs to be there, and trickle down to the derivative products.  Using this linked override system would eliminate the current issue where every single variation of a `Filament Setting` or a `Print Setting` for a printer, needs to have a unique name and duplicates every single setting when you may only need to change only 1 setting.

legend069, 2023-11-27:
Redesigning the GUI will be avoided.
There are so many users that like the design/layout. And they would get mad if it gets changed in a substantial way.

But please feel free to draw something up that would project what you have in mind. That would help us understand better.

I can see a fan modifier override being usefull.
But not extra `folders` and changes for the GUI.

Kaelum, 2023-11-27:
@legend069 as I stated previously, the original idea for this is useless and would be a complete waste of time to implement.  I'll probably submit this idea for change to PrusaSlicer, as it doesn't make any sense to make this type of change here, which I also stated previously.

closed 2023-11-28 by Kaelum

Kaelum, 2023-11-28:
For those interested, I created the following issue under PrusaSlicer:

https://github.com/prusa3d/PrusaSlicer/issues/11776