supermerill/SuperSlicer #3938: Arc overhangs and when the next version will be out?

laptop-laptop, 2023-09-22:
**Is your feature request related to a problem? Please describe.**

Hello, it seems that the latest version was released a year ago. How come there is no update since then? Am I missing something?

If possible, could you lease add Arc Overhangs?

freighttt, 2023-09-24:
You aren’t missing anything. The creator had to take a break, as life happened and they were a solo dev working for free. Recently they announced they were starting up again and would be working on susi full time thanks to some corporate interest. There was a new nightly about a month ago, and it is stable. As it stands, susi is still hands down the most capable slicer for those needing precise control, only missing a couple blingy features which will absolutely be merged to susi soon. If you had searched the other issues, you’d have seen this question asked a million times. Susi is still alive!!!

j-snijder, 2023-09-25:
Where to find this susi  nightlys then ?

mk-maddin, 2023-09-26:
> Where to find this susi nightlys then ?

https://github.com/supermerill/SuperSlicer/releases

j-snijder, 2023-09-26:
Ah, okay thats the same repo 😊

 

https://github.com/supermerill/SuperSlicer/releases

Dit e-mailbericht is met Avast-antivirussoftware gecontroleerd op virussen.
www.avast.com

rcarmo, 2023-10-05:
I don't see any nightlies there - or anything that is dated after March 4

mk-maddin, 2023-10-05:
Thats fully correct - don't get your issue here.
"2.5.59.2 - Mar 04 23" is the latest beta / nightly
"2.4.58.5 - Sep 24 22" is the latest stable.

rcarmo, 2023-10-05:
I dug around and the actual nightlies in the Build tab are also not updated for a while now (March for Mac ARM). So maybe add a note for that someplace...

reloxx13, 2023-10-19:
and github actions nightly have expired

supermerill, 2023-11-22:
will be after ps merge. in the coming weeks