supermerill/SuperSlicer #3942: Compiling under MacOS creates incomplete or damaged dmg file

Lausestrick, 2023-10-01:
### What happened?

I followed the instructions to build on MacOS, which run and create a dmg file which leads to a damaged or incomplete dmg file. I copied the dmg into my application folder, which does not do the trick. 

[image]

During the compiling process, there are some errors showing, which seem to originate from CMake. I reinstalled this software, but same result. Tried it several times, very reproducible. 

I am puzzled and need some expert advice how to get this to run on my machine. Many thanks in advance!

This is my build.log file:
[Build.log](https://github.com/supermerill/SuperSlicer/files/12778494/Build.log)

### Project file & How to reproduce

1. git clone https://github.com/supermerill/SuperSlicer.git
2. cd resources/profiles
git submodule update
3. ./BuildMacOS.sh -i

### Version

I cannot open super slicer

### Operating system

MacOS Sonoma 14.0 - 8-core Intel i9

### Printer model

Voron 2.4 with Stealthburner

erathos77, 2023-10-05:
I  am not expert but I successfully built SuperSlicer 2.5.59.2 on macos 11.6 /XCode13.2 following a minor change in src/qhull/CMakeLists without error. Your build.Log is strongly similar to mine, except : 
- ressources file size: 40.6Mo vs 72.8Mo for mine, 
- The last "pack" line is missing at the end before the .dmg: pack/SuperSlicer/SuperSlicer.app/Contents/MacOS/SuperSlicer in mine. That probably means  the SuperSlicer binary file is missing in your SuperSlicer/contents/MacOS folder, i.e. your built has completely failed and will never run on your machine.
Instead of the build.log, please analyze the complete compilation/debug process obtained in the terminal: there are surely several errors reported (not the warnings), such as library not found or directory error...
If you find something failed about qhull as me, I can help you.
However, I don't know whether source.ss2.5.59.2 would be macos Sonoma compatible. Why not use the github .dmg  release to try before?

Lausestrick, 2023-10-08:
Many thanks for the proposal, however I could not get it to work. I tried the dmg from GitHub and it works now!!! THANKS!

closed 2023-11-20 by supermerill