supermerill/SuperSlicer #3944: Make internal bridge controls able to be independent of default bridge controls

Salmonsquire, 2023-10-03:
When printing large surfaces, I am having difficulty in dialing in values for the internal bridge layer that don't also impact the external bridges. The result is that there is poorer top surfaces, with occasional pillowing or overextrusion. The current method to resolve this is to lower the bridge width and bridge density. This reduction improves the first solid infill layer.

This would affect all printers

The solution I propose is to make it so that internal bridges are able to dialed in independently of default/external bridges. (Possibly even additional control for the interaction between internal bridge and the first solid infill layer)

legend069, 2023-10-21:
internal bridges and normal bridges are basically the same.
the main difference between the two is internal bridges gets "supports" in some areas from infill.
since the infill density will always change. it's not worth having the two extrusion roles separated just to tune them. and then have another model get screwed up from having a different infill area supporting the internal bridge infill.

Salmonsquire, 2023-10-21:
You can have normal bridging dialed in well and then interaction between the top infill layers and the internal bridging can lead pillowing. This is repeatable. There is also a short thread on OrcaSlicer for the same issue.

supermerill, 2023-11-20:
If you dial your external bridges, what is the problem with your internal bridges?
can I have some picture of the end result (with both good external bridge and problemtic top layer due to internal bridge)? and the project used to slice it?

Salmonsquire, 2023-11-20:
Hey @supermerill  (Responding from mobile)

I was chasing this problem. 
https://media.discordapp.net/attachments/773406440664727592/1168753518719553587/20231030_232716.jpg
https://cdn.discordapp.com/attachments/773406440664727592/1151158422268555405/20230912_073831.jpg
https://cdn.discordapp.com/attachments/773406440664727592/1168729010021339176/20231030_185851.jpg
I was reducing the bridge flow to compensate for the weird build up of material. I was able to get a smooth top surface by reducing that flow. 

Ultimately I discovered that my problem stemmed from cooling being too strong/efficient. I don't know if settings are still needed at this point for the other requesters, but I wonder if there is a bridging test for dialing in cooling would be appropriate.

closed 2023-11-20 by Salmonsquire