supermerill/SuperSlicer #3947: execve failed: No such file or directory

synfinatic, 2023-10-06:
### What happened?

Trying to "upload and print" the g-code and getting an error that "execve failed: No such file or directory"

Sadly, doesn't say which external program it is trying to exec?  I assume this has something to do with the upload of the gcode file.  

Additionally, I can say that I installed SuperSlicer on a new ARM Mac and imported the "export config bundle with physical printers" from my old Intel Mac which was working fine before.  So I assume this has something to do with ARM vs. Intel on the Mac platform.
[SuperSlicer_config_bundle.ini.zip](https://github.com/supermerill/SuperSlicer/files/12825817/SuperSlicer_config_bundle.ini.zip)

### Project file & How to reproduce

[Body1.3mf.zip](https://github.com/supermerill/SuperSlicer/files/12825820/Body1.3mf.zip)

### Version

2.4.58.5

### Operating system

macOS 13.5.2 / ARM

### Printer model

Voron 2.4

synfinatic, 2023-10-06:
Might be related, but I'm also using https://github.com/Annex-Engineering/klipper_estimator as a post processing script. seems to be working fine on my Intel Mac, but ARM Mac is getting the same execve error.

synfinatic, 2023-10-07:
Oh, this error seems to be happening with the post processing script, not the upload.  I just deleted my post-process script and it uploads just fine.

File permissions and all that seem to be correct on the `klipper_estimator` binary and it runs fine by hand.  Not sure how to debug further.

supermerill, 2023-11-20:
On mac arm, are the scripts totally broken or are they oaky?
Can you try to execute some simple program like python with a hello world?

synfinatic, 2024-01-28:
python works just fine on my system (installed via brew).   Not sure what else I can check or how to debug this?  Is there some way to know what is being executed?