supermerill/SuperSlicer #3948: MacOS Startup Crash v2.4.58.5 - missing libzstd.1.dylib file

iAmVaeVictus, 2023-10-06:
### What happened?

1. Download SuperSlicer_2.4.58.5_macos_220924.dmg. Open and copy into Applications folder. 
2. Attempt to open, immediate crash.  
3. Apply 'xattr -cr /Applications/SuperSlicer.app' in terminal window. 
4. Attempt to open, immediate crash. 

Crash report says 
**Crash Report** 
Process:               SuperSlicer [24878]
Path:                  /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Identifier:            com.merill.slic3r/
Version:               SuperSlicer SuperSlicer_2.4.58.5_2022-09-23 (SuperSlicer_2.4.58.5_2022-09-23)
Code Type:             X86-64 (Native)
Parent Process:        launchd [1]
User ID:               502

Date/Time:             2023-10-06 11:22:45.0415 -0500
OS Version:            macOS 13.6 (22G120)
Report Version:        12
Anonymous UUID:        558C1CD0-014C-80E4-43E4-2A9A93F87FE5

Sleep/Wake UUID:       AAB1D897-256C-4643-A43F-0C7144E008D2

Time Awake Since Boot: 260000 seconds
Time Since Wake:       5590 seconds

System Integrity Protection: enabled

Crashed Thread:        0

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace DYLD, Code 1 Library missing
Library not loaded: /usr/local/opt/zstd/lib/libzstd.1.dylib
Referenced from: <EA3F7209-F390-3268-9C90-D8081DCEDC30> /private/var/folders/*/SuperSlicer.app/Contents/MacOS/SuperSlicer
Reason: tried: '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/opt/zstd/lib/libzstd.1.dylib' (no such file), '/usr/local/lib/libzstd.1.dylib' (no such file), '/usr/lib/libzstd.1.dylib' (no such file, not in dyld cache)
(terminated at launch; ignore backtrace)

### Project file & How to reproduce

N/A 

### Version

2.4.58.5

### Operating system

MacOS 13.6 Ventura

### Printer model

Voron 2.4

erathos77, 2023-10-06:
Try 2.5.59.2 version; [releases /Arc fitting edition/Assets] : SS_2.4.59.2_macos_230304.dmg  (it is 2.5.59.2 in reality)
Also you can try in terminal if you had not done that before: 
brew update
brew install zstd

iAmVaeVictus, 2023-10-09:
Thanks, will do.

<https://moes.com/>

Scott Portis

Owner - 9 Middle TN Locations

Cannon Restaurant Management

d/b/a Moe's Southwest Grill

***@***.***

(615) 383-8993 <6153838993> (w)

(615) 300-3696 <61553003696> (c)

P.O. Box 159029
Nashville, TN 37215

moes.com

<https://www.facebook.com/MoesMiddleTN>

<https://twitter.com/MoesNashville>

<https://www.instagram.com/moes_nashville/>

<https://locations.moes.com/tn>

Rak3tenmann, 2023-11-03:
same here.
Same error log.
Is there a little bit more context on 

brew update
brew install zstd

Not sure what to do.
THanks!
Christian

Rak3tenmann, 2023-11-04:
The Intel OSX Version was hard to find and hidden beding a "wrong" version-number here:
https://github.com/supermerill/SuperSlicer/releases/tag/2.5.59.2
;-)
However - thanks a lot. This one works.
Cheers
Christian