supermerill/SuperSlicer #3952: Crash with "ERROR_FILE_TO_SEND_TO_MERILL_PLZZZZ"

TheRaf974, 2023-10-24:
### What happened?

I just found this file "ERROR_FILE_TO_SEND_TO_MERILL_PLZZZZ.txt", so there it is. 
I only remembered SuperSlicer crashed a few days ago, don't remember the exact situation. 
Not 100% sure but may have happened while using one of the calibration utilities.

Sorry, don't have more information.

### Project file & How to reproduce

[ERROR_FILE_TO_SEND_TO_MERILL_PLZZZZ.txt](https://github.com/supermerill/SuperSlicer/files/13120464/ERROR_FILE_TO_SEND_TO_MERILL_PLZZZZ.txt)

### Version

2.4.58.5

### Operating system

Windows 11 Pro 22H2

### Printer model

Ender 3

Creo2005, 2023-10-25:
use the orca slicer. SuperSlicer is already dead. Forget about him :)
orca is now the best slicer and is developing.
script for transferring settings https://github.com/theophile/SuperSlicer_to_Orca_scripts

smcbride-ca, 2023-10-25:
> use the orca slicer. SuperSlicer is already dead. Forget about him :)

https://github.com/supermerill/SuperSlicer/discussions/3902#discussioncomment-6883183

supermerill, 2023-11-20:
Thank you.
Seems like there is some case when it's created by mistake.
If you didn't have any problem with fill pattern beeing replaced unexpectedly by a rectilinear, it's all good. sorry.

closed 2023-11-20 by supermerill