supermerill/SuperSlicer #3954: Feature Request Multiple buildplate + Travel Time calculation

Meszaross, 2023-10-29:
Please implement, multiple build plate like in orca/bambu slicer. This is a huge help for bigger project when you need to slice lot of object. Helps for easier calculation for summary material need and print time. 
Please also implement Travel Time display (time for travel moves needed)

mlewis-everley, 2023-11-07:
+1 For multiple build plates

j-snijder, 2023-11-07:
+1 from me alsoSent from Android device

supermerill, 2023-12-14:
What would be a good way to implement it in the gui?
If it's on the old layout, with one tab per platter, I guess I can use a tab per platter.
If it's on one of the new layout with no /multiple tabs per platter, maybe adding a new tab row?

BioCookieYT, 2023-12-14:
> What would be a good way to implement it in the gui? If it's on the old layout, with one tab per platter, I guess I can use a tab per platter. If it's on one of the new layout with no /multiple tabs per platter, maybe adding a new tab row?

Would it be possible to do it in the Orcaslicer way? I think it would be the best approach

BioCookieYT, 2023-12-14:
> I think a tabbed approach might be a good way. I find Bambu/Orca's approach can get a bit annoying if I have more than 4 plates.

Yeah but you have a good overview of the plates, with a tabbed approach you would have to go through every tab when searching something

mlewis-everley, 2023-12-14:
True, if there was a way to show/hide plates at least, that would probably be a good compromise?

mlewis-everley, 2023-12-14:
Or maybe if only the render view was tabbed, and the right project overview panel showed content from all tabs. That was you could select the object you are looking for and maybe it automatically switched to the correct plate in the rendered view?

Meszaross, 2023-12-14:
> What would be a good way to implement it in the gui? If it's on the old layout, with one tab per platter, I guess I can use a tab per platter. If it's on one of the new layout with no /multiple tabs per platter, maybe adding a new tab row?

like in bambu/orca works please

dragoonmc, 2023-12-14:
I like Orca's way of doing it and being able to slide object between plates. but you would also need to include the ability to lock a plate so that auto arrange doesn't try to cram all of the plates together again

supermerill, 2023-12-23:
So more somethign like:
 - a big build plate, with the grids / textures in the right places
 - a new arrange algo & out-of-plate detection that works on cut sections
 - the slicing algorithm works on cut sections sequentially, and output a gcode per section.

new settings: 
- build plate count per x & per y
- distance between build pate

did i miss something?

BioCookieYT, 2023-12-25:
> **- a big build plate, with the grids / textures in the right places**
 
Yes, I think OrcaSlicer handles this similar, a big build plate with textures in the right place. Would it be possible to make the texture between the plates transparent so it looks similar to OrcaSlicer?

> **- a new arrange algo & out-of-plate detection that works on cut sections**
> **- the slicing algorithm works on cut sections sequentially, and output a gcode per section.**

Yes

> **- build plate count per x & per y**

Would it be instead doable to add a button in the UI called "Add Plate", where the Slicer adds automatically a new build plate? 
[image]

> **- distance between build pate**

Yes, good idea!

> **- did i miss something?**

- buttons like "Arrange", "Lock plate" or "Delete" next to the build plate 
- Titles for the build plates 

**But first enjoy your holidays and take a break from developing <3**

supermerill, 2023-12-29:
> Would it be instead doable to add a button in the UI called "Add Plate", where the Slicer adds automatically a new build plate?

So just a build plate count, and I arrange them automatically in a square?

BioCookieYT, 2023-12-29:
> So just a build plate count, and I arrange them automatically in a square?

yes

TheAnimatrix, 2024-09-11:
Is this coming ?