supermerill/SuperSlicer #3958: Custom G-Code - "current_position" - incorrect?

dumitrugrl, 2023-11-09:
### What happened?

Using the current_position inside different custom G-Code sections does not seem to return the expected tool position

I was expecting them to return the coordinates of the previous (or last) G1 line before evaluating the variable. Is this the right assumption?

At a tool change, I am trying to compare the current X with a minimum limit allowed (stored in another variable) to avoid a crash before moving towards zero.

Any suggestions on how to achieve this?

I noticed that the variable is set using "**unscaled**()" function, but I am not sure what that does.

X shows -117.304, while I was expecting 64.482 (from the last move executed).

[image]

### Project file & How to reproduce

N/A

### Version

2.5.59

### Operating system

windows 10

### Printer model

Creality CR-10S

dumitrugrl, 2023-11-09:
I don't know if the unscaled values are used some place else, but making the changes below (in the mentioned method) returns the expected values.

**src\libslic3r\GCode.cpp**  -> `std::string GCode::placeholder_parser_process`

`
default_config.set_key_value("current_position", new ConfigOptionFloats({unscaled(m_last_pos.x()), unscaled(m_last_pos.y()) }));
        
Vec3d vpos = this->writer().get_position();
double x_ = vpos.x();        
double y_ = vpos.y();        
default_config.set_key_value("current_position",new ConfigOptionFloats({x_ , y_}));   // <- this gives the g-code coordinates
`

supermerill, 2023-11-20:
mmm
First i though of using  `point_to_gcode(m_last_pos)` 
Then I guessyour idea of using directly the gcodewriter position is safer.
For what i have checked, it's only updated in the same thread right now.
I added the z as it's available this way.
Also added current_object_position to have the position of the current object. `current_position-current_object_position` should get you the old value (if there is no extruder offset)
thank you for the report & fix.