supermerill/SuperSlicer #3959: Add support for Elegoo .goo format

Duckle29, 2023-11-15:
**Is your feature request related to a problem? Please describe.**
I'd like to be able to use one slicer for all my printers, and so far it seems SuperSlicer can do it.

**Describe the solution you'd like**
I'd like SuperSlicer to be able to export in the .goo format [foss spec](https://github.com/elegooofficial/GOO/blob/main/Goo%20Format%20Spec%20V1.2.pdf) directly, but an alternative solution could be to use UVtools to convert the sl1 file.

My original plan was to invoke UVTools through a post processing script, but these don't seem to be available for resin printers for some reason. The benefit of doing that is that it'd support a lot more formats than just .goo

**Describe how it would work**
Option one: Native goo
In the setup wizard, a "custom resin printer" option would be present, and you could set up a printer here that exports to .goo format.

**Describe alternatives you've considered**
In the setup wizard, a "custom resin printer" option would be present, with the option to export using UVTools. Not sure how you'd prefer to package that dependency, or have people enter a path to the binary.
there'd be options for output format, and perhaps any options that are in the output format, that wouldn't be present in the sl1 format, but I'm not familiar with the file types for that info.

A similar issue to this may be here: https://github.com/supermerill/SuperSlicer/issues/1488
The UVTools approach might be able to solve that too.

supermerill, 2023-11-20:
I've plan to add other output format as gcode, but will take a bit of time to have the framework needed.

Duckle29, 2023-11-20:
Sounds dope. Any hints on where to start looking if I'd want to add the post processing step in an SLA printer profile? Or am I better off just manually yoinking the file through UVTools for now? :)

supermerill, 2023-11-21:
for dlp resin, it's easier.
There is already two output format: SL1 and CWS
each are in their files, in `src/libslic3r/format/`
you can look at both. There is only one method to fill: `export_print`.
in the input, you have the settings, the layers (a list of images) and you have to write that in the file.

This system is only useful to lcd printer.
sadly, prusa named their dlp as "sla" even if there's no laser.
there is currently no sla/sls workflow. I won't make one myself unless i'm paid for, as i don't have a printer for it and so will only do a subpar job, unable to test.

Duckle29, 2023-11-24:
Cool! Thanks. I might poke at it in the weekend 😊