supermerill/SuperSlicer #3961: Hole Perimeter Count - An important Function missing in all slicers.

AstroZYX, 2023-11-17:
While I’m working on optimize the printing plan of a design, I realize an important function is missing in all the slicers, search confirm this, lot of people need the exact same function in a different way:

Hole Perimeter Count / Hole wall line count

Plus, “Hole size definition”, to defind a size, bigger than this will be consider "outside", smaller will be considered a hole

A picture explan the problem:

[image]

I was able to get away by using “Close Holes” & 0% infill. In a design with consistent XY Dimension. Like the one on the far left of the image, the line didn’t break, able to form the strongest and cleanest part, also less movement, travel and seams.

Normally this is impossible due to the hole inner surface been considered as “Outer Surface” of the model.

But this method no longer works on a tapered design, it will make the hole become tapered too. 

A lot of people need this function is because they want to reinforce the bolt hole only, not the entire model, this usually end up with walls been added to entire model then print failed due to warp, happen to me few times, the modifier only make the model even weaker so it is not a solution (the far right is one example)

Can you consider adding this function in a future version?

Thank you very much for all your efforts.

supermerill, 2023-11-17:
I don't understand the issue.
In the left, the space is closed, with a bottom and a top.
In the middle, there is no bottom nor top, so it's the exterior, a visible surface.
I don't understand the right.

AstroZYX, 2023-11-19:
Thanks for the respond and sorry for the confusion bring by the image, I create another one:

[image]

The two on the left is how parts normally sliced, Due to hole wall count can only be the same as outside wall count, the part is significant weaker tha, the middle one

The one in the middle is the one I want, to archive this, I have to Close all Hole set 0% Infill, than use line width and spaceing to adjest the diameter of the hole.

This method not working on taperd (or size & shape changes while height increase) design since the hole shape follow the outer shape.

Many reason led to the part been designed like this (form the struture without any infill), and we need thousands of it, this is the 1st try to use 3D printing to replace inject molding completely, and we really don't want this to fail because of a missing funcation in slicing softwares.

I know there maybe algorithms to detect holes on models, since you can make it bigger, smaller, delete it, why can't we have a funcation let us define how many walls we need around a hole?

Hope this can be solved in near future.

Thank again

supermerill, 2023-11-20:
The problem imho, is from the 'tear'-shaped hole that is added to the middle one (comparing to the left one.
This 'hole' can't be guessed, even more with a taper. 
I can do a 'please no inner perimeter, but it will be more or less the same as your workaround, the difference is that you won't need to adjust manually the number of perimeters for the first layers.
Can you send me your project? So i can send you the modified version that will be the same as with the 'new' feature?

AstroZYX, 2023-11-20:
Hi Merill

Thanks for the fillowup 

This is the project file for referance, we do need an editable file/version for future design change & fine turn the hole size for different materials.

[Hole Problem.zip](https://github.com/supermerill/SuperSlicer/files/13418005/Hole.Problem.zip)

Not sure how to archive......this funcation may not mean anything for people print toys, but it can be crucial for funcatioal parts printing.

let me try bring up examples show lots of situation is funcation can be a life changer.

With in these models, the only area need extra walls is the hole, in some model, we don't want any infill structure in between the outer walls and the holes, some other we want the hole connect to each other,  the parts no longer printable by ABS or ASA due to warping if we add wall to the entire model:

[image]
[image]

Cooldown warping will ruin everything

[image]

supermerill, 2023-11-21:
here is the project with the midle one modified so no perimeters 'from the external' go inside a hole.
[Hole Problem_modified.zip](https://github.com/supermerill/SuperSlicer/files/13420668/Hole.Problem_modified.zip)
I think it's possible to do that.

Now, in your last message, i think you talk about something else: allow to set external & hole perimter separatly.
I already tried to implement that but it's not easy. Tthere is so many weird edge case... also a hole can morph into the external skin, and vice versa. I think i'll try again in the future, when i'll rewrote this part.