supermerill/SuperSlicer #3962: Klipper test connection fail (ss trying connection to octoprint?)

sedovserge, 2023-11-18:
### What happened?

If printer works on Klipper and choose klipper.

SS try to connect to octoprint when klipper choose and test connection is fail.
This bug issued in clean run (without settings):
[image]

Meanwhile in version 2.4.58.5 test connection is ok:
[image]

### Project file & How to reproduce

Printer settings > Add phisical printer > set klipper connection > press test

### Version

2.5.59.2

### Operating system

windows 10

### Printer model

Custom

supermerill, 2023-11-20:
Can you verify that if you check the `enable 2-way ssl`, there is nothing in the `client certificate file`.
That's mostly the only change between the two, and a possible gui malfunction.

sedovserge, 2023-11-20:
> Can you verify that if you check the `enable 2-way ssl`, there is nothing in the `client certificate file`. That's mostly the only change between the two, and a possible gui malfunction.

The same problem
[image]

DaTTcz, 2023-11-24:
When I tested it, I got this: 
Nelze se připojit k Klipper: Timeout was reached:
Connection timed out after 10003 milliseconds
[Error 28]

sedovserge, 2023-11-25:
v2.5.59.3 Resolved! Tnx

closed 2024-02-11 by supermerill