supermerill/SuperSlicer #3964: Add Auto-Calc option for Wipe towers

BioCookieYT, 2023-11-20:
Hi, first of all thanks a lot for coming back to SuSi merill. Hope it's not too difficult to catch everything up. 

Over the last time MMU printers became more popular and Slicers like Bambu Studio integrated features to make MMU printing easier. One of those features is auto calculation the purging volume based on the color. So for example if you switch from a white color to a dark one you need less filament compared to switching from black to white. From my testing the auto calculation works pretty reliable and can save you a lot of filament. It would be great to see such feature in SuSi :p 

[image]

supermerill, 2023-11-27:
set the `pigment percentage between 0 and 1`
then activate `single extruder mmu` (you should already have it selected)
then in the printer->single mmu setup, you can activate `advanced wipe tower purge volume calculus`
choose the settings 
i'll paste the tooltip here:
```
"Linear : volume = nozzle + volume_mult * (pigmentBefore-pigmentAfter)\n"
 "Quadratic: volume = nozzle + volume_mult * (pigmentBefore-pigmentAfter)+ volume_mult * (pigmentBefore-pigmentAfter)^3\n"
 "Hyperbola: volume = nozzle + volume_mult * (0.5+pigmentBefore) / (0.5+pigmentAfter)"
```

BioCookieYT, 2023-11-27:
Oh I didn't know this was already a thing in Superslicer. It seems like OrcaSlicer has some way to calculate the pigment percentage based on the color which makes the whole process even more easier. Could you please check if this would be possible also for SuperSlicer? Would be an awesome feature for the MMU community :)

BioCookieYT, 2023-11-27:
Not an expert in this topic but I think the calculation happens here 
[image]
https://github.com/SoftFever/OrcaSlicer/blob/main/src/libslic3r/FlushVolCalc.cpp