supermerill/SuperSlicer #3965: Arachne plus classic slicing combined.

NiiXXiiN, 2023-11-23:
**Is your feature request related to a problem? Please describe.**
No, it's a quality of life - esthetic request 

**Describe the solution you'd like**
Often times I need to slice projects with the arachne engine due to faster printing and better mechanical properties of the result but don't do it because classic slicing has better looking top and bottom surfaces. 
This would create new possibilities for complicated models. 

**Describe how it would work**
Chexkboxes or drop down menus allowing for one type of slicing for specific number of layers on top, bottom or both and another type of slicing for the rest. 
Or like color switching - extruder selection with mmu after initial slicing with the + icon on the layer selection in gcode viewer to switch style of slicing. Like adding custom gcode on specific heights the ability to change slicing engine. 

I really don't know if it's doable though.

supermerill, 2023-11-23:
basically, allow the switch to be on a modifier (like the height range modifier), right?

NiiXXiiN, 2023-11-23:
Something like that. If that what it's called when you modify at different heights the color or extruder.
What do you think, wouldn't this be a nice feature?

DaTTcz, 2023-11-24:
I would look for it in the Height Range Modifier... Add setting... Perimeters Shell... arachne/classic. Maybe dropmenu, or switch checkboxes.