supermerill/SuperSlicer #3966: How to download the nightly for windows?

ganzgustav22, 2023-11-23:
Sorry if this is a stupid question. Did the Github UI change or something?

Instructions say:

> Nightly builds are available through the [git actions page](https://github.com/supermerill/SuperSlicer/actions). Click on the build for your platform and then on the archive name (nightly_win64 or SuperSlicer-gtk2.AppImage or nightly_macos.dmg) in the "Artifacts" section in the top right corner.

I click on the actions page, then click on "C/C++ Nightly Windows x64" on the left side.

Then I'm here, but there is no archive name and no "Artifacts" on that page (?)

[image]

Pavulon87, 2023-11-23:
Select one of the lasts runs with green checkmark.
[image]

And download archive nightly_win64
[image]

nick9one1, 2023-11-23:
I've noticed that you [nightly_win64] button to be clickable, otherwise it's just text.

Pavulon87, 2023-11-23:
I'm not sure, but you probably have to be logged in to be able to download.
Try this link: [nightly_win64](https://github.com/supermerill/SuperSlicer/suites/18422624630/artifacts/1067919977)

DaTTcz, 2023-11-24:
I confirm that it can be downloaded without problems. I tried nightly_win64 and rc_win64.

ganzgustav22, 2023-11-25:
Thanks a lot, I was just too blind to see :)

And shortly after I ask how to download the nightly, there is a new release, finally. Really nice to see how this project picks up speed again, thanks a lot to supermerill.