supermerill/SuperSlicer #3969: Toolpath outside print area

Calcousin55, 2023-11-25:
### What happened?

After slicing there can be seen a toolpath outside print area was detected in gcode preview.
[image]

### Project file & How to reproduce

[idler_bearing_cover.zip](https://github.com/supermerill/SuperSlicer/files/13465105/idler_bearing_cover.zip)
Using the default Voron V2 300mm3 0.4mm nozzle profile with 0.10mm layer height
Using the following settings causes toolpath error
Skirt & Brim -> Draft shield -> Enable
Infill -> Sparse -> 40% Cubic

### Version

2.5.59.3

### Operating system

Windows 10

### Printer model

Voron V2

supermerill, 2023-12-23:
It's because if you use complete_objects and brim_per_object, the brim needs to be shifted. But it shouldn't if no complete_objects.

Calcousin55, 2023-12-29:
acorreya's issue looks to be fixed in 2.5.59.4 but my issue still persists in 2.5.59.4 but looks like its fixed in 2.5.59.5

closed 2023-12-29 by Calcousin55