supermerill/SuperSlicer #3973: "Layer time goal" does not work.

Murrdo, 2023-11-26:
Version 2.5.59.3

Windows 10

"Layer time goal" does not work.
I tried to set any time and speed values ​​- it did not affect the slising.

[image]
[image]

supermerill, 2023-11-27:
Sadly, the layer tiem goal doesn't take into account the acceleration right now, but tthe time calculation done byt the gcode viewer afterward do.
So the time used by the cooling tab is very optimistic.
It's somethign that can be improved upon, but imo it's best to first reworj the gcode process. Currently, the gcode string are created too soon, and it should be the last things to do, as it created these weird situation where the fan & speed adjustments are really done by post-processing scripts.

Snelso91, 2023-11-27:
Just thought I'd add my +1 to this issue. I have the same problem in 2.5.59.3, and it's not present in 2.5.59.2 or 2.4.58.5 from what I can see.

Darkmann12, 2023-11-29:
+1 here too. I thought I was going crazy. Figuring this out is also what led me to find out that the Cooling tab in previous versions crashes now, too

Snelso91, 2023-11-29:
@Darkmann12 speaking of going crazy, I had the exact same issue with the filament cooling tab crashing on previous versions after I opened 2.5.59.3! Took me forever to figure out how to fix it and it was a pretty heavy-handed method:

1. Close superslicer.
2. Make a copy of the AppData\Roaming\SuperSlicer folder if you don't have a backup of it already from before you launched 2.5.59.3.
3. Delete the AppData\Roaming\SuperSlicer folder.
4. Launch an old version such as 2.4.58.5 to have it recreate the folder with its defaults
5. Then copy the filament, physical_printer, print and printer folders and SuperSlicer.ini file from a backup from before you launched 2.5.59.3 back into the new superslicer appdata folder. If you don't have a backup, use the copy of the folder that I mentioned in step 2, but not sure if that will work compared to the clean appdata folder that hasn't been touched by 2.5.59.3.
6. Launch old superslicer and you should be able to access the cooling tab now, and you should have kept you original filament, printer, etc. profiles.

It was hard to nail this down because I went systematically file by file in the entire superslicer appdata folder comparing the backup versions from 2.4.58.5 to the current appdata folder after launching 2.5.59.3. And so as I went along, I restored most of the lines in the files to original state (using file compare in VSCode) that I thought might have been the cause (such as the superslicer.ini or print profiles), but this still didn't fix it, even after I changed back all the non-relevant lines too like `# generated by SuperSlicer 2.4 on 2023-11-27 at 19:43:24 UTC`. And even weirder than that was that I eventually changed almsot all the files back to the same as the backup and it hadn't fixed it, so I literally copied the entire superslicer appdata folder from the backup and overwrote the current one with it, and somehow that didn't fix it either...
So it seemed the key step was to fully delete the superslicer appdata folder and have old superslicer recreate it with the defaults, and then restore the pieces of it that you needed from backup (such as filament, print and printer profiles and the ini file).

Also fyi, I had to revert back to 2.4.58.5 instead of 2.5.59.2, because I tried 2.5.59.2 but it also had a different critical bug which was that if you enable vase mode it crashes superslicer.

agzorb1842, 2023-12-15:
@supermerill Are you saying it actually does work, but just not in the time estimation in the slicer? Or have I misunderstood you?

pinwc4, 2023-12-21:
I am encountering this problem as well and had to revert back to 2.5.59.2 which slows downs print speed as expected to maintain a minimum layer time goal.  2.5.59.3 does not slow it down at all.  I was able to revert by deleting the ui folder, deleting all the configs is not needed.

Snelso91, 2023-12-22:
Yeah just deleting the UI folder could be the easier solution. After I had copied back the entire app data folder, and it still didn't fix it (which was very confusing given that that exact same app data was working before the change), I basically gave up trying to be nuanced then and just deleted the whole appdata folder as a last resort which I wasn't expecting to work but did.

There must be something very strange going on when deleting the UI folder works, but restoring the UI folder (and all the other appdata) to a known working backup state does not work.

supermerill, 2023-12-23:
> @supermerill Are you saying it actually does work, but just not in the time estimation in the slicer? Or have I misunderstood you?

yes. It's using a crude time estimation, not the reliable one.

agzorb1842, 2023-12-23:
> > @supermerill Are you saying it actually does work, but just not in the time estimation in the slicer? Or have I misunderstood you?
> 
> yes. It's using a crude time estimation, not the reliable one.

Unfortunately that doesn’t seem to actually be the case for me. I will do some more experimenting as I am only about 80% certain, though. 

Thanks!

Darkmann12, 2024-02-14:
Do we have a timeline for when we might get this fixed, @supermerill?

agzorb1842, 2024-02-14:
> I didn't found any problem on the algorithm each time I was given a project file. The only improvement is to better compute the time so it won't have the little difference between the setting & the final output. 

That is quite confusing for me as it definitely does not slow down the layer time — ie it is not  an incorrect time estimation any of the times I have used it. I have had to switch slicers for most things as a result, which is unfortunate since SS has otherwise been my go-to…

Darkmann12, 2024-02-14:
> That is quite confusing for me as it definitely does not slow down the layer time — ie it is not an incorrect time estimation any of the times I have used it. I have had to switch slicers for most [...]

100%. I've found myself putting speed modifier boxes around smaller layers because it just does not care... it is a royal pain

agzorb1842, 2024-02-14:
Well, at least I know it’s not just me.

supermerill, 2024-02-14:
Give me an exemple project.

Darkmann12, 2024-02-14:
> Give me an example project.

OK I tested using a different computer and it appears to be no longer broken - updating SuperSlicer to the latest release (*.8) fixed the issue for me. It could be worth saying that newer versions don't have completely broken layer time goal, as it is not explicitly said in any of the newer releases than the affected version (*.3). @agzorb1842 Have you updated your SuperSlicer yet? I know I held off because I didn't see any note of it being fixed in the changelogs, and was waiting to update until then.

supermerill, 2024-02-15:
> It could be worth saying that newer versions don't have completely broken layer time goal, as it is not explicitly said in any of the newer releases than the affected version (.3).

probably this changelog line
[image]

I should ahve write a better description (with layer time goal and such), sorry

Darkmann12, 2024-02-15:
> > It could be worth saying that newer versions don't have completely broken layer time goal, as it is not explicitly said in any of the newer releases than the affected version (.3).
> 
> probably this changelog line ![image](https://private-user-images.githubusercontent.com/6536403/305016353-8ddff5e0-2579-4f39-969b-e9126d029758.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc5ODg5NjUsIm5iZiI6MTcwNzk4ODY2NSwicGF0aCI6Ii82NTM2NDAzLzMwNTAxNjM1My04ZGRmZjVlMC0yNTc5LTRmMzktOTY5Yi1lOTEyNmQwMjk3NTgucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMTVUMDkxNzQ1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9M2VkNmM5OWNiOTkzNDVkMTFmM2M4NDY1ZTMxZDBmZjYxOWY4YTgxZmJkOTc2YTcxMGE0NDMwMTQyN2VhM2U1MSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.P48-6M9VpAfrLf1uaxaVKsZ6S2p1FT_lDppc1kLAi9c)
> 
> I should have written a better description (with layer time goal and such), sorry

No worries, I completely missed that myself. All is right in the world haha

agzorb1842, 2024-02-16:
> OK I tested using a different computer and it appears to be no longer broken - updating SuperSlicer to the latest release (.8) fixed the issue for me. It could be worth saying that newer versions [...]

I hadn’t even noticed there was a new version. I’ve now updated and it seems to be working. 

Thanks!