supermerill/SuperSlicer #3975: [2.5.59.3-win64] Gcode Preview Vertical Slider Legend spacing/crowding

Alohacajun, 2023-11-27:
**Bug description**
The legend for the vertical slider in Gcode preview is not properly fitting in the window. This is exacerbated with taller models.

**Expected behavior**
The legend should be scaled according to the number of layers in the file. This worked properly in 2.5.59.2

**To Reproduce**
Steps to reproduce the behavior:
1. Load any 3D model and slice it. Tall models exaggerate the behavior.
2. Open Gcode Preview
3. Vertical bar legend spacing error appears

**Screenshots**
 Example of the broken spacing:
[image]
How it should look:
[image]

**Desktop:**
 - OS: Win64
 - Version 2.5.59.3
 - Voron 2.4_350

supermerill, 2023-12-23:
can't reproduce.
I added a big box and the spacing of the text is good.
Can you share a project?

Alohacajun, 2023-12-24:
> can't reproduce. I added a big box and the spacing of the text is good. Can you share a project?

I've attached a zipped folder with a default cylinder scaled in Z-axis. I also provided the config via "Export Config". There's also another screencap of the resulting error in there as well.

[expected_error.zip](https://github.com/supermerill/SuperSlicer/files/13762851/expected_error.zip)