supermerill/SuperSlicer #3976: 2.5.59.3 causes older versions to crash when viewing cooling settings

Alohacajun, 2023-11-27:
**Fix: Delete %AppData%/SuperSlicer/ui_layout**
**Bug description**
If you run 2.5.59.3 and switch back to an older version (ie 2.5.59.2), opening Filament > Cooling will cause it to crash.

**Expected behavior**
A clear and concise description of what you expected to happen.

**To Reproduce**
Steps to reproduce the behavior:
1. Run 2.5.59.3, then close.
2. Run 2.5.59.2
3. Open Filament tab
4. Open cooling options
5. Crash

**Desktop (please complete the following information):**
 - OS: Windows
 - Version 2.5.59.3 / 2.5.59.2

**Additional context**
_This might happen because all my versions share the same AppData config folder._
Is this related to the "Rework of fan speed control by extrusion type." in 59.3?

legend069, 2023-11-28:
Yes this is most likely due to the fan speeds rework.

Not sure why it would crash though, unless it's unhappy with extra variables in the config.

Alohacajun, 2023-11-28:
Would it be viable to have alpha/beta/RC builds use their own AppData directory? They do this in the PS repo as seen [here](https://github.com/prusa3d/PrusaSlicer/releases?page=2).

Darkmann12, 2023-11-29:
I have also run into this :)

legend069, 2023-12-10:
@supermerill 
would it be worth while reverting the variables that were changed back to original?
or adjusting code so any old variable names are still loaded from the config but get assigned to the new variable names?
ie `overhang_perimeter_fan_speed  = overhangs_fan_speed`

also https://github.com/supermerill/SuperSlicer/issues/3991

supermerill, 2023-12-20:
The issue comes from the ui layout.
The layout used is the last one, but with an old version, there can be empty lines, where the settings doesn't exists yet.
And currently, there is a bug with a crash when there is an empty line.
I fixed it, and next version will have the patch.
I'll patch the 2.5 and future 2.7
For the 2.4, I'm not sure it's that useful, tell me.

To manually fix the issue: before using an old version, delete the ui_layout directory from your configuration directory.
The way it's updated is really subpar, i'll work a bit on that right now to avoid future problem with it.

closed 2024-02-11 by supermerill