supermerill/SuperSlicer #3980: Artefacts. Perimeter lines in between the part.

mypaxa00, 2023-11-30:
### Version
2.5.59

### Operating system type + version
Windows 10 Enterprise 22H2

### Behavior
I have wierd artifacts after slicing with my model
Actual Results:
[image]
[image]

#### Project File (.3MF) where problem occurs
[Battery_Pack_Cap_No_Supports.zip](https://github.com/supermerill/SuperSlicer/files/13513868/Battery_Pack_Cap_No_Supports.zip)

Murrdo, 2023-12-01:
+1 
[image]

drone777, 2023-12-01:
Got such artefacts many times too.
+1

neophyl, 2023-12-02:
Its basically this one https://github.com/supermerill/SuperSlicer/issues/3785

Alohacajun, 2023-12-03:
As described in [this ](https://github.com/supermerill/SuperSlicer/issues/3785#issuecomment-1556030404)response of #3785 it is related to overhang generation. Unchecking overhangs completely fixes the issue in your model. If you still need overhang generation, a temporary fix is to reduce the "Bridge speed & fan threshold".

mypaxa00, 2023-12-03:
Duplicate
https://github.com/supermerill/SuperSlicer/issues/3785

closed 2023-12-03 by mypaxa00