supermerill/SuperSlicer #3985: SuperSlicer failed to run in win11

paul73211, 2023-12-03:
### Version
2.5.59.3
### Operating system type + version
Win11

### Behavior
Crash at start up.
[image]
[image]

supermerill, 2023-12-19:
#3988

paul73211, 2023-12-21:
I am using an Intel i7-12800HX with an NVIDIA RTX A3000 32Gb GPU. After copying opengl32.dll from the mesa folder to the root directory I got Superslicer running ok with a slowly speed on slice and preview process.

legend069, 2024-02-12:
I've had this issue running SS inside my VM

SidheQn, 2024-03-16:
Just downloaded this and opened it with no problem on Win 11 Pro. Sliced a crystal flexi dragon and it took maybe 1 minute.

Perhaps the version of Windows 11 matters. Home and Home S may be problematic

Felix14-v2, 2024-08-07:
Also crashes on my machine 🥲
After the first message it shows this internal error:
[image]