supermerill/SuperSlicer #3988: Superslicer exits with error when starting

gerthalberg, 2023-12-04:
### What happened?

Starting Superslicer results in the application stooping with an exception. copying opengl32.dll from the mesa folder to the root directory stops
[image]

### Project file & How to reproduce

Start Superslicer without the opengl32.dll present

### Version

2.5.59.3

### Operating system

windows 11 Udgave	Windows 11 Pro Version	23H2 Installeret d.	‎14-‎12-‎2022 Operativsystemets build	22631.2715 Serienummer	PF49DQCM Visning	Windows Feature Experience Pack 1000.22677.1000.0

### Printer model

Custom - corexy - Klipper

supermerill, 2023-12-19:
I guess i'll have to isntall windows 11

gerthalberg, 2023-12-19:
Pls. let me know if I can help or otherwise assist. It puzzles me, that it logs, that it's unable to parse the version of the AMD driver. I'm not running any AMD devices It's an Intel CoreI9 HX and an NVIDIA RTX A2000 8Gb GPU. It's a Lenovo Thinkpad P16 with an 3840x2400 OLED display.

supermerill, 2023-12-23:
I think it's an opengl graphic dirver issue with win11.
I guess prusa already fixed it, we'll see.

gerthalberg, 2023-12-23:
Have just downloaded PS 2.7.1 and yes, it runs - so I'll wait until you merge down the road.

closed 2023-12-23 by gerthalberg