supermerill/SuperSlicer #3989: Creality Thumbnails

eff3ry, 2023-12-05:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Creality printers expect thumbnails in a different format to what is provided by superslicer.
It expects a form of JPG file in the gcode but the default jpg option in superslicer does not seem to work so i am forced to use PNG with a post processing script.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Impliment a "Creality JPG" option for thumbnail format possibly 
[image]

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Ive been using the post processing python scripts from here [https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview](https://github.com/mriscoc/Ender3V2S1/wiki/How-to-generate-a-gcode-preview) analyzing this script may have insight into the format that creality printers expect

Thanks for your time, let me know if there is anything I missed.

supermerill, 2023-12-23:
Can you give me a SuSi project file with a simple little cube, the gcode you get from it, and the gcode from the script, that works?
I can do it myself, but it will save me some precious time.

eff3ry, 2023-12-24:
Hi thanks for the response i hope the following files are helpfull

[Gcode and Project Files.zip](https://github.com/supermerill/SuperSlicer/files/13760864/Gcode.and.Project.Files.zip)
### Before post processing Gcode
[image]
### After post processing Gcode
[image]

eff3ry, 2023-12-26:
Well, I did a little testing and it seems that generating the gcode with JPG selected and then in the gcode renaming `thumbnail_jpg` to just `thumbnail` seems to let the thumbnail load fine so the only problem I'm really having is that my printer expects `thumbnail` while superslicer generates `thumbnail_jpg` 🤦‍♂️