supermerill/SuperSlicer #3992: Build for ARM Mac

ssombra, 2023-12-07:
Hi, I recently moved to Mac and I want to install a newer version of Superslicer. I see that there are not builds available for ARM macs. How can I get one?

DasBurninator, 2023-12-08:
> Thanks for the answer. Do you mean the "Arc Fitting Edition"? That is the last one I see with a Mac build.

The latest nightly build. Under actions... Not the latest release.
https://github.com/supermerill/SuperSlicer/actions/runs/6961460900

ssombra, 2023-12-08:
> > Thanks for the answer. Do you mean the "Arc Fitting Edition"? That is the last one I see with a Mac build.
> 
> The latest nightly build. Under actions... Not the latest release. https://github.com/supermerill/SuperSlicer/actions/runs/6961460900

Thank you very much! As you can see I am not very familiar with Github :) I really appreciate your support.

Hr46ph, 2023-12-09:
On a similar note, I asked this yesterday:
https://github.com/supermerill/SuperSlicer/discussions/3995

I tried to build the ARM version which completes, but is unable to start.

supermerill, 2023-12-14:
Sorry for that, but I'll review the arm build after I have a good 2.7 version