supermerill/SuperSlicer #3996: Creating a physical printer silently fails when the printer name contains invalid path characters

NameOfTheDragon, 2023-12-09:
### What happened?

When creating a physical printer, with name "Voron V2/1360", all seems to work, except that the next time the program is loaded, the printer is gone.

### Project file & How to reproduce

- Create a physical printer named "Voron V2/1360" - this apparently succeeds and the printer can be used.
- Open the configuration folder under Help -> Show Configuration Folder
  It will be observed that the physical printer ini file has not been created.
- Close and restart SuperSlicer
  It will be observed that the physical printer is gone.

### Version

2.5.59.3

### Operating system

Windows 11

### Printer model

Voron 2.4/350 with Stealthburner and Clockwork 2

supermerill, 2023-12-14:
In the 2.7, there is some checks about that.
I'll see if I can find some times to port that back to 2.5

Adams-Scott, 2025-02-01:
I was having the same issue in 2.9. I was naming my physical printer ".6". If I named it "Prusa 6" it was fine. I'm guessing it had to do with the decimal.