supermerill/SuperSlicer #3998: Independent linear/pressure advance for internal and external perimeters.

MartijnGevaert, 2023-12-11:
**Is your feature request related to a problem? Please describe.**
Bulging on seams, especially for thick layers.

Bulging on seams is reduced by tuning linear/pressure advance. However, these values are tuned for a specific speed, acceleration, ... Specifically, it's recommended to tune these for the external perimeter settings. This means that the linear advance values are not tuned for the internal perimeters.
Most of the time this is not an issue, but on thick layer lines (0.4mm layer w. 0.6 nozzle), the bulging of the internal perimeters pushes out the seam of the external perimeter.
The more internal wall perimeters, the more obvious this issue should become.

**Describe the solution you'd like**
I believe if the internal perimeters could have a different PA value, tuning could be done separately for internal and external perimeters, having more filament being put down with it's optimal PA value..

**Describe how it would work**
 
The slicer would simply have to emit a gcode value whenever it switches between internal/external perimeters.
There might be even cooler implementations where a PA curve is used for different speeds... but that's getting a bit out there.

**Describe alternatives you've considered**
There are other solutions regarding flow around the seems, such as stopping earlier, coasting, ... although they work in practice, they are more of a patch on underlying differences between the modeled flow and actual flow.

**Additional context**
None for now, let me know if there is any additional investigation/information which I can add which would be of use here.
Unfortunately I can't test the separate PA values in practice yet because it's not trivial to add these gcode commands manually as far as I know.

MartijnGevaert, 2023-12-13:
Amazing! I'll run it on a couple of prints. Any specific testing you're looking for that I can contribute?

legend069, 2023-12-13:
> Amazing! I'll run it on a couple of prints. Any specific testing you're looking for that I can contribute?
> Also looks like it must have taken forever to tune in all those values. Where you able to derive some from other values?

There's nothing really to test, aslong as you edit my snipit appropriately it should work flawlessly for you.

Yes it did take a hour or two to get the PA values.
You only need to run a PA test if a speed/acceleration value is different than 50mms/5000

Example; internal perimeter ER is 400mms and 30k Accel.
External perimeter ER is 200mme and 10k accel.
You would want two PA values.

If your other ER changes alot you would need a PA values for each.

MartijnGevaert, 2023-12-13:
Work's great, especially great for solid infill as I'm now seeing less bulging since I can tune my PA for separate cases better and much more!
Only thing that could come out of this ticket now is a different UI experience, but in my opinion a programmatic style like the G-code you provided is probably the best way.
Feel free to close if you agree.

legend069, 2023-12-20:
I'll start working on this when the new release is finished.
Will add a PA calibration tool the same method as [this](https://ellis3dp.com/Pressure_Linear_Advance_Tool/)
Will have it pull speed,width, acceleration values automatically.

User would just need to select what ER to generate a test for and slice it then print it.

MartijnGevaert, 2023-12-21:
That would be amazing, anything that helps speed up tuning in new filaments is welcome. Would also be very cool if multiple ER's could be combined in a single calibration print.
Simply printing one after the other, but not having to clean bed and start new print in between would be a big time saver.

MartijnGevaert, 2024-01-04:
I have an idea that is very much related to this issue, and might be the most elegant solution to it, which I have not seen in any other slicer yet.

@supermerill you added a post showing your PA value for different accelerations.
You modified 2 parameters, PA and smooth time, but for the sake of my explanation I'll simplify it to a single parameter, PA.

Let's assume that you always run the same smooth time, and you run a PA test for following accelerations:
1000, 2500, 5000, 10000
I think it would then be very likely a curve start showing that can be used to determine the correct PA value for all other accelerations in between.
This would drastically reduce the amount of calibration prints needed.

Adding in a second data point, smooth time, would then simply make it a 3d graph, which would require some more data points, maybe 5 or 6 + some assumptions on what the graph looks like (linear/exponential/..) to get good estimations of non-tested combinations.

## Testing the concept
This process can already be done manually today to verify if there is an underlying predictable PA behavior which can be taken advantage of, let me know if you think this idea makes sense.

## Feature value
If this works, it would mean that Superslicer can pick the right PA for every single acceleration, and require no re-tuning if you modify your accels, such as is common for different qualities (quality/draft), and layer heights

## Dreaming ahead
- If it turns out almost all toolheads follow the same graph shape, it might be possible to get the amount of required calibration data points down to only 3 (for PA only).

- This would also perhaps allow the incorporation of more parameters, such as layer height, by simply adding more calibration points and increasing the graph dimensions.

legend069, 2024-01-04:
> That would be amazing, anything that helps speed up tuning in new filaments is welcome. Would also be very cool if multiple ER's could be combined in a single calibration print.
> Simply printing one [...]

Adding several models for the single plate should be fine, currently the code wipes the plate each time you make a model, easy enough change to make.
Simple "check box/button" to clear plate should do the trick.

I'm wanting to add in a option that would make a model for all ER and it will make several lines just so the user can easily identify what needs adjusting.
In short, this would be a "pre test" to make tuning it easier. Although I'm not sure if I should make a model for each letter/word and add that too next to each pre test line

legend069, 2024-01-04:
> Adding in a second data point, smooth time, would then simply make it a 3d graph, which would require some more data points, maybe 5 or 6 + some assumptions on what the graph looks like (linear/exponential/..) to get good estimations of non-tested combinations.

Aswell as individual boxes for each ER value, I'm also wanting to add exactly what you've mentioned. A kind of " auto adjustment for PA, but the math for this one could be a bit tricky to implement.

I've still got a bit of work to do for the PA calibration tool to be finished.i might have to make the GUI injection part for PA at another time.

MartijnGevaert, 2024-01-04:
I'm not sure the PA math would be all that complicated it's essentially just fitting a graph to data points.
I'm sure there's a bunch of c++ libraries that have graph fitting functions available, maybe even some that are already part of the dependencies.
I think it's very likely that the best fitting graph will always be an inverse exponential graph.

I can do this test:
calibration print for PA at 500, 1000, 2500, 5000, 7500, 10000.
Get a fitted graph for each combination of 4 data points and see how close the graphs are to each other (closer=better).
If the graphs are very close, it means it's possible to estimate the other data points using a graph based on 4 calibration points.
I can do this test maybe later today, feel free to comment if you have some ideas/doubts about this testing methodology in the meantime.

supermerill, 2024-01-16:
If you want a fast calibration print, maybe you can print just a line in "L" shape (one for each acceleration).
then by looking at the start, corner and end, maybe you can decide if it's good, under-pa (start late, corner bulge, end ooze) or over-pa (start blob, thin corner, missing end).

NiiXXiiN, 2024-01-18:
I just want to say that I was about to post a request for this feature exactly "variable pressure advance for different speeds and accelerations". I am really glad to see that there is already interest in this feature.
Such an amazing community!

MartijnGevaert, 2024-01-18:
I did some calibrations and I do think there is underlying behavior that can be taken advantage of, however, I did it with a 0.6 nozzle thinking it would be easier to spot PA, but it actually made it harder, I don't think for example the 7000 and 10000 acc values make sense, but for those there is much less difference between 0.01, 0.02... they all look the same.

I'll redo the test with 0.4 nozzle in the future.
[image]

legend069, 2024-01-31:
> I did some calibrations and I do think there is underlying behavior that can be taken advantage of, however, I did it with a 0.6 nozzle thinking it would be easier to spot PA, but it actually made [...]

you'll find that when printing faster and faster you need to start changing the ``smooth_time`` variable.
on a 'large' enough model, and dependent on the accuracy of the calipers you should be able to measure the ``widths`` of the lines when running a PA test.
i have not done any real world measurements/data gathering. but in my hypothesis, in the picture below area '2'/'4,'5 is where the smooth time takes place. but since the acceleration values are so FAST! the 'smoothing' of the extrusion actually covers the distance (example) of 5mm, instead of 0.1mm compared to a lower acceleration value.
with a larger 'smoothing' distance it's a result of harder to tell PA values apart.
[image]

i recently just started printing with 50k accelerations ~40mm³/s speeds and can confirm the same 'very little visible differences' when looking at ``advance`` numbers alone. i did notice width changes when changing it to smooth time though.

MartijnGevaert, 2024-01-31:
AFAIK know, smooth time should be kept as low as possible, but is introduced to avoid asking the extruder to do things which it can't do, such as very fast retract and deretracts.
That means smooth time should be a constant, across all accels, which is good because that means it's one less dimension to the issue.
This means we are left with speed and accel.
I still need to do the tests again at different speeds to see if there's a predictable 2D-plane in there.
That being said, @legend069 your input is useful as it can explain why my 7000 and 10000 accel tests where difficult to read.

legend069, 2024-02-17:
i've partially completed the PA calibration tool https://github.com/supermerill/SuperSlicer/pull/4115

on the topic of adding in PA values that would be injected into the gcode
theres a few ways i've considered doing this

1. base value for PA in filament settings tab. ie; PA=0.025 
then in the print settings tab would be where the adjustments would be, box for each ER role with the user adding a modifier to the base value for PA +/-
(but this will soon turn issues with using the same print settings for different filaments)

2. having the PA values only in the filament settings tab. the user would need to have several filament profiles per print settings speeds/accelerations.(i assume the would already since more speed = different part cooling)

i'm thinking the best place for them is possibility 2
to save the user switching back and forth when entering in the PA values it should be possible to add a "tooltip" type window.
(similar to the part cooling description) that just lists the speeds/accelerations