supermerill/SuperSlicer #3999: Random Support Layer Above Printed Model

Super-Matter, 2023-12-13:
### What happened?

[image]
[image]

### Project file & How to reproduce

[ASIC Miner Shroud v13.zip](https://github.com/supermerill/SuperSlicer/files/13660262/ASIC.Miner.Shroud.v13.zip)

supermerill, 2023-12-19:
can't reproduce with 2.5.59.3 or my latest dev, or the nightly of the date you have.
It's totally random?

Super-Matter, 2024-02-12:
[image]
Still happens on 59.6 

But changing offset value from "from plane" to "from filament" fixes it for some reason 
[image]

Actually seen some weird behavior of "from filament" offset value "negative flow generated" errors

I'll try to find a recent example of this in a form of a 3mf project

Super-Matter, 2024-02-12:
The config snapshot from the bug above 
[20240212T092833Z.zip](https://github.com/supermerill/SuperSlicer/files/14236803/20240212T092833Z.zip)

Super-Matter, 2024-02-12:
[image]
generated Goode asks printer to draw it midair and then ends print normally

supermerill, 2024-02-12:
still can't replicate with the .3mf from the top.