supermerill/SuperSlicer #4002: Missing features for QIDI X-MAX 3 printer

EddyCoubard, 2023-12-16:
### Version
_Version of SuperSlicer : 2.5.59.3

### Operating system type + version
_Windows 10 Professionnel - 22H2  - Build 19045.3803 - French

### Behavior
* _Describe the problem_

Owner of a Qidi X-MAX 3 3D printer, I am starting to use SuperSlicer which is much more powerful and interesting to me than the proprietary QidiSlicer slicer.
To use SuperSlicer with this printer, I retrieved the PrusaSlicer settings from Qidi here: https://drive.google.com/drive/folders/1-T0OQ0e-pdG6GqgNiTiRoGWq26b_cIEI
On the other hand, Super Slicer is missing some interesting features, namely:
The management of Fans no. 2 and 3 (see image Qidi_Slicer_Fan2+Fan3.png from QidiSlicer),
Pressure Advence management (see image Qidi_Slicer_Pressure_Advance.png)

Do you think these features could be added to the next version of SuperSlicer, or is there already a possibility of having them?

I'm not a regular Github user, I hope I made this request correctly.

* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_
 * _Screenshots if useful_

_Is this a new feature request?_
Related guides for writing feature requests: http://meta.stackexchange.com/a/259196 http://nickohrn.com/2013/09/write-great-feature-request-bug-report/

#### Project File (.3MF) where problem occurs
_*Upload a Project File* (.3MF) `Plater -> Save Project as...`_
_You have to zip it (or rename it to .zip) to be able to drag& drop it here_
[image]
[image]

neophyl, 2023-12-16:
Given that the screen shots of Qidi slicer looks like its based on Prusa Slicer then the chances of the pressure advance stuff being available is pretty damn good.  Merill is working on merging in the Prusa Slicer 2.7 stuff at the moment I believe.

So most/all of the 2.7 extras should make it in SuSi.  It looks lime Qidi added in  the cooling chamber fan speed though so that one is more of an unknown.  However Merill is pretty good about evaluating user requests :)

d-kozminihs, 2024-09-12:
Is it possible to get the same results via custom filament G-codes?