supermerill/SuperSlicer #4003: Add option not to apply Fill Gap settings to the first and last layer or make it as separate settings.

Lajbror, 2023-12-17:
It would be very helpful to be able to set different values to Gap fill on bottom and top layers.

For example if I my preference is not to fill the gaps that are shorter than 1mm and have surface smaller than 1mm2, then this is applied also in the bottom and the top layer, which is not desired behavior because it leaves unfilled gaps in those visible layers and the print does not look as good as it should.

What I do now is I make Layer height modifier and set Gap fill values separately for the top and bottom layers. But doing so for every print is tedious.

This would be very helpful feature in my opinion.

supermerill, 2023-12-19:
for first layer thing:
https://github.com/supermerill/SuperSlicer/issues/412

for the "top layer", there is no "top layer".
There is a "last layer" and "top areas". My guess is to have something for "top areas"

Lajbror, 2023-12-19:
Yea, sorry for the incorrect terminology. But I guess you get the idea. The thing is, many models do not need to fill all the gaps through the entire model so my settings is not to fill the gaps that are shorter than 1mm and have surface smaller than 1mm2, which saves some time and some other relating problems. But what is needed is to fill the very bottom layer and the very top layer which are the layers that are visible and these layers need to be filled correctly. (Maybe you call it differently and I am sorry again if I do not get those terms correctly, but I really hope you get the idea and I am sure this would be just great new feature supporting or completing existing feature in the slicer :-)

EmanuelKroell, 2023-12-19:
I love this idea! This could save quite some print time for a lot of models I'm printing

Lajbror, 2023-12-19:
To [EmanuelKroell](https://github.com/EmanuelKroell)
Hi Emanuel, you might have misunderstood my new feature request. The possibility not fill the gaps by your needs is even now possible in Superslicer. Go to "Print settings" -> "Perimeters & Shell" -> in "Advanced" section look for "Gap fill threshold" and set it to your preferences. It is very useful feature. The problem is, that it applies to all layers, from the bottom to the top, but you do not want it to be applied to the first and last layer, which are the surfaces that you can see. You want these layers to be properly filled. And this is my New feature request -> Not to apply these settings to these visible layers.