supermerill/SuperSlicer #4004: Walls printed one layer height higher when using semi-paralell printing

Lecso11, 2023-12-17:
### What happened?

When printing multiple part with semi paralell printing, every part except the first has the walls printed one layer height higher (in my case 0.2mm) than it should be printed.
My first layer is 0.25mm, others 0.2. First part's first layer is perfect, but the other's walls are printed on 0.45mm height, then the infill on 0.25. 
On the picture: left is first, rest are identically printed wrong.

### Project file & How to reproduce

[kampov3.zip](https://github.com/supermerill/SuperSlicer/files/13697565/kampov3.zip)
[image]

### Version

2.5.59.3

### Operating system

windows 10

### Printer model

Voron 2.4 klipper

vovodroid, 2023-12-18:
I opened project, but it's configured to be printed by layer

[image]

What are your settings here?

Lecso11, 2023-12-18:
> I opened project, but it's configured to be printed by layer [...]

Hi!
Weird... Here they are:
[image]

vovodroid, 2023-12-18:
> My first layer is 0.25mm
0.3 actually - 75% of 0.4 nozzle. 

> others 0.2.
You have variable layer height, so it varies.

I don't see such issue. Second layer of all parts is printed at 0.505:

[image]

[image]

But actually despite putting efforts and time in this feature development, I'm now thinking that it's not necessary. I was initially thinking about stringing and printing time, but eventually stringing is not issue for me, and printing time also is almost the same. And it's better to print small objects by layer to allow better cooling.

Lecso11, 2023-12-18:
> I don't see such issue. Second layer of all parts is printed at 0.505: [...]

Not the second, the first layer's walls are printed on wrong height. But now that I thought about it, that 0.2mm higher, which results in the 0.45 might be my z-hop.
See: 
[image]

Actually this part of the infill is still too high:
[image]

But this is fine:
[image]

What you say is partly true, but I think it is still a very useful feature. While you might have no issue with stringing, there are materials that are pretty bad. And with middle-sized parts, or where cooling is not an issue, it still makes sense to use. 
I think the feature has its use-case, I wanted this for a long time.

vovodroid, 2023-12-18:
> I don't know what happened, but you pretty much cannot see my settings.

I downloaded SS release, but didn't run config wizard, I guess all settings should be included in project file, isn't it?

> there are materials that are pretty bad.

I've read about it. But I printed PLA, PETG, ASA, Nylon, Polycarbonate and TPU. TPU indeed is stringing. Other - nothing special.

Lecso11, 2023-12-18:
> I downloaded SS release, but didn't run config wizard, I guess all settings should be included in project file, isn't it?
>
>
> I've read about it. But I printed PLA, PETG, ASA, Nylon, Polycarbonate and [...]

Yes the projekt file should contain it.

Anyway, if you find the time looking into it, I would be thankful. I would like to use this feature.

vovodroid, 2023-12-18:
> the project file should contain it.

So something strange here. I loaded you projects and all setting are default:

[image]

Lecso11, 2023-12-18:
> > the project file should contain it.
> 
> So something strange here. I loaded your project and all setting are default:
> 
> ![image](https://private-user-images.githubusercontent.com/8691781/291254706-69d7099e-58aa-41ac-b293-7ddaac17e34f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDI5MjE0NzUsIm5iZiI6MTcwMjkyMTE3NSwicGF0aCI6Ii84NjkxNzgxLzI5MTI1NDcwNi02OWQ3MDk5ZS01OGFhLTQxYWMtYjI5My03ZGRhYWMxN2UzNGYucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQUlXTkpZQVg0Q1NWRUg1M0ElMkYyMDIzMTIxOCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyMzEyMThUMTczOTM1WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YmM0NmQ0N2I4Yjk3MjNlNTk2MjYzMDdjN2E1OTU4ZWEwOTdlNjY0MTAzMTZiYTVhNmFjNDU1MjY2ZTE1YzZhYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.41UYPYS0VIU-fb5tBxpHOBRDPKU1uljrnrw_uiRyF-w)

[kampov3forgithub.zip](https://github.com/supermerill/SuperSlicer/files/13706835/kampov3forgithub.zip)
try this one, It works for me. Loads everything.

vovodroid, 2023-12-19:
They should be separated objects:
[image]

And full config is still not loaded:
[image]

Can you try open project in fresh SS installation, better on other machine?

supermerill, 2023-12-19:
note: i can reproduce, and can open your project on latest nightly.

supermerill, 2023-12-19:
Fixed.
That was a quite serious one.
The bug originates from the part that remove superfluous feedrate ( `F3000`). In this case, it's a wrong check: it has to check the line to remove the whole of it if it's only a `G1 F3000`, but not if it's `G1 X1 Y2 F3000`. The check was completely insufficient.

Lecso11, 2023-12-22:
> Fixed. That was a quite serious one. The bug originates from the part that remove superfluous feedrate ( `F3000`). In this case, it's a wrong check: it has to check the line to remove the whole of it if it's only a `G1 F3000`, but not if it's `G1 X1 Y2 F3000`. The check was completely insufficient.

Thank you!

closed 2024-02-11 by supermerill