supermerill/SuperSlicer #4010: Seam position doesn't show up in search

orangetruth, 2023-12-21:
### What happened?

Seam position doesn't show up in the search under any of the terms "seam", "seam position", or even the variable name "s_seam_position"

### Project file & How to reproduce

From 3D View tab (or any of the settings tabs), hit ctrl + f and search for any of the terms "seam", "seam position", or "s_seam_position". You won't find s_seam_position anywhere

[image]
[image]
[image]

### Version

2.5.59.2

### Operating system

Windows 10

5axes, 2023-12-21:
Also in 2.5.59.3

supermerill, 2023-12-23:
already done in the nightly, but still need a bugfix.

closed 2024-02-11 by supermerill