supermerill/SuperSlicer #4011: Can't Slice Files

mlee12382, 2023-12-21:
### What happened?

When slicing files in 2.5.59.2 or 2.5.59.3 I receive the following error.
Failed to rename the output G-code file from C:\Users\mlee1\AppData\Local\Temp\.21660.gcode.tmp.postprocess to C:\Users\mlee1\AppData\Local\Temp\.21660.gcode.tmp
Is C:\Users\mlee1\AppData\Local\Temp\.21660.gcode.tmp.postprocess locked? (gcp)permission denied

Also If I try to use custom variables it gives the following:
SuperSlicer has encountered a fatal error: "Access (status) violation"
Please save your project and restart SuperSlicer. We would be glad if you reported the issue. 
Please join the 'crash_log.txt' file (if it exists) content to your report.

I can slice normally with 2.5.59.0 though it does seem to have some strange behavior when adding supports.

### Project file & How to reproduce

Slicing any model.

### Version

2.5.59.2 and 2.5.59.3

### Operating system

Windows 10

### Printer model

N/A

supermerill, 2023-12-23:
Can you put your project file here?

mlee12382, 2023-12-23:
[crash_log.txt](https://github.com/supermerill/SuperSlicer/files/13760025/crash_log.txt)

Here's the crash log, the error is not project dependent, I open 2.5.59.(2/3) and add a basic cube to the build plate and try to slice it and it throws the above mentioned errors. Not sure if they are related to each other or if the custom variable issue is something completely separate.

The goofy stuff with the supports on 2.5.59.0 seems to happen whenever I use supports on any model. They don't seem to actually affect the print. I slice the file and then open the gcode in the viewer app and they show up, they are not visible in the sliced preview in the slicer however. When I slice the same file with supports in PrusaSlicer the model appears to look normal.

supermerill, 2023-12-23:
can you try this nightly? https://github.com/supermerill/SuperSlicer/actions/runs/7308766355/artifacts/1132647650
If the error continue to happen with it, i'll create a debug one to be able to have a better crash log.

mlee12382, 2023-12-23:
[crash_log.txt](https://github.com/supermerill/SuperSlicer/files/13760167/crash_log.txt)

Unfortunately the errors still persist. 
When using a my custom Spoolman ID variable I receive the following:
SuperSlicer has encountered a fatal error: "Access (status) violation"
Please save your project and restart SuperSlicer. We would be glad if you reported the issue. 
Please join the 'crash_log.txt' file (if it exists) content to your report.

When I remove the custom variable from the start gcode I receive the following:
Failed to rename the output G-code file from C:\Users\mlee1\AppData\Local\Temp\.5232.gcode.tmp.postprocess to C:\Users\mlee1\AppData\Local\Temp\.5232.gcode.tmp
Is C:\Users\mlee1\AppData\Local\Temp\.5232.gcode.tmp.postprocess locked? (gcp)permission denied

mlee12382, 2023-12-30:
So, interestingly, it seems to be related to the custom variables. I downloaded 2.5.59.5 and tried it and was observing the same behavior as with .2 and .3. Before I first tried to sliced a model with .5 though I removed my custom variable from the start gcode and added a basic shape cube to the build plate and it sliced just fine. When I then put the custom variable back into my start gcode it gave the Access status violation, I then removed the variable and tried again and received the postprocess locked error. 
Closing SS and restarting and then removing the variable again allows me to slice models. So there appears to be some conflict with custom variables.

supermerill, 2023-12-31:
if you can paste here your project, i would have been able to tell that to you.
And if you want a fix, it's still needed.

mlee12382, 2023-12-31:
[cable_frame_anchor_x6.zip](https://github.com/supermerill/SuperSlicer/files/13802251/cable_frame_anchor_x6.zip)
Here is a project file.

supermerill, 2024-01-04:
I fixed how it handle the variable. Now it should consider it as a string.
note: in the meantime, you may be able to use it with quote:
`spoolman="SPOOLMAN_ID=21"`

mlee12382, 2024-01-04:
Great! I'll give that a shot.

mlee12382, 2024-02-08:
I realized that I didn't need to have the whole string in the custom variable. All I really need is spoolman=<id> in the notes and then I can do my call by doing SPOOLMAN_ID={spoolman} not sure why I didn't see that sooner.

closed 2024-02-14 by supermerill