supermerill/SuperSlicer #4012: Rectilinear grid support pattern is not grid

Lecso11, 2023-12-22:
### What happened?

Setting the support pattern to grid results in the same pattern as regular rectilinear.
[image]
[image]
Expected (from PrusaSlicer):
[image]

### Project file & How to reproduce

[prt0001.zip](https://github.com/supermerill/SuperSlicer/files/13753207/prt0001.zip)

### Version

2.5.59.3

### Operating system

windows 10

### Printer model

Voron 2.4 klipper

supermerill, 2023-12-23:
what i have on prusaslicer 2.5:
[image]
I didn't succeed to replicate your image. Do you have a prusaslicer 2.5 project?

The first layer bottom is only present on snug, i'll look at why.

Lecso11, 2023-12-23:
> what i have on prusaslicer 2.5: [...]

Hey, looking at the picture, you probably have supports only on the bed checked.

I use Prusa 2.7.0. Prusa project file: 
[support_test_PS.zip](https://github.com/supermerill/SuperSlicer/files/13759685/support_test_PS.zip)

I would also have a question. The first layer of the part, that is printed on supports are "bridge infill". In SS these are printed in 45°, while in PS 0°. With a rectange 45° results on longer lines, which is harder to bridge. The behaviour is the same without supports, which is the more problematic, as these are true bridges. My question is, did I miss a setting in SS, or this is the way it slices bridges? I noticed, that SS slices bridges in an inefficient way.

supermerill, 2023-12-23:
The current version of supereslicer is at 2.5 version, the 2.7 isn't ready yet.
You can't compare a 2.5 with a 2.7 and expect the same result.