supermerill/SuperSlicer #4021: Bridge infill will not stick to the overhang perimeters when low Infill/Perimeter encroachment values.

Lajbror, 2023-12-27:
When the Infill/Perimeter encroachment is set to 15%, it also affects the bridge infill overlap, causing it not to adhere to the overhang perimeters due to minimal or no overlap. As far as I know, there isn't an option to set the bridge infill overlap separately.

The setting "Anchor solid infill by X mm - Bridged" doesn't have the intended effect, it performs differently from what's needed.

Many models encounter this issue, and there isn't a straightforward method for setting this. It occurred to me to rotate the bridge infill to ensure proper overlap at all necessary points, but in this case, there's no use for it. This limitation persists, especially when using low values for Infill/Perimeter encroachment, which is essential for my requirements.

It would be incredibly helpful for users to be able to adjust this overlap according to their specific use cases.

15% Infill/Perimeter encroachment: If I want to use 15% Infill/Perimeter encroachment, it does not stick and bridging infill sags.
[image]

Default setting Infill/Perimeter encroachment 25% is much better and there is no issue
[image]

supermerill, 2023-12-27:
When creating perimeters, the fill areas are created. I can see the overhangs giving a bit more area to infill.

Lajbror, 2023-12-27:
That would certainly help. I made some further inspection and I can certainly say that the lower the value of "Infill/Perimeter encroachment" is the worse the bridge area adheres to overhangs and bridge infill sags. In my particular case values 18% and lower does not adhere at all. Maybe it would be good idea to take into account, that the lower the Infill/Perimeter encroachment value is, the worse the bridge infill adheres to the overhang and the more it is needed to compensate for it.

On the other hand if you will not make it user adjustable, there might be some other issues, for example if you lower the values of "Max line overlap" or "Solid fill overlap" but I have not tested this yet.

Anyway it is totally up to you and I will be happy for any adjustments you make.

And just so you know, in tooltip you say to report usefulness of the feature, so here it goes:
The settings "Print settings -> Width & Flow -> Perimeter overlap" is very useful feature making the external perimeter objectively smoother in some cases", so I definitely vote for not removing it. I use all three values "Default", "External" and "Gap fill".

supermerill, 2023-12-27:
I wonder if a different encroachment value for internal bridge can be enough?

Lajbror, 2023-12-27:
Yes, but just to be clear, this is not a problem of an internal bridge infill but the problem of bridge infill :-)