supermerill/SuperSlicer #4022: [FR] Merge single top wall for Arachne

vovodroid, 2023-12-27:
Hi, I recently ported it from Orca to Prusa  [Port single wall top feature for Arachne from Orca](https://github.com/mjonuschat/PrusaSlicer/pull/2)

I guess it's easy to take it to SS.

supermerill, 2023-12-27:
seems good.
I'll have to review my function for split_top_surfaces from the classic perimeter, to merge into one. Both seems the same, but there may be some enhancement or difference, i guess.

vovodroid, 2023-12-27:
> I'll have to review my function for split_top_surfaces from the classic perimeter,

Any meaning of supporting classic generator? Cura now has only Arachne.

supermerill, 2023-12-27:
Some printer may not like the variable width.
It's easier to modify to test things.
On some prints, I prefer the pathing of the classic one.

vovodroid, 2023-12-28:
@supermerill It seems you took not latest code, there is updated version https://github.com/mjonuschat/PrusaSlicer/pull/2/files

neophyl, 2023-12-28:
Many of us printing still prefer classic to Arachne.

supermerill, 2023-12-28:
> @supermerill It seems you took not latest code, there is updated version https://github.com/mjonuschat/PrusaSlicer/pull/2/files

hmm, I forgot a chunk, yes.

vovodroid, 2023-12-28:
Not only chunk, https://github.com/mjonuschat/PrusaSlicer/pull/2/files#diff-5ff8894b878e5ceb707151bd13f0a10c22a06fb202d959f428422596e0defb75R1760 produce better result then original calculation.

supermerill, 2023-12-28:
If it's just
 ` coord_t offset_top_surface = scale_t(config->external_infill_margin.get_abs_value(peri_count == 0 ? 0. : max_perimeters_width));`
vs
` coord_t offset_top_surface = ext_perimeter_width;`
then you can always set it manually, it's a setting after all.

I discover a little oversight: in split_top_surfaces, they use the perimeter nozzle diameter instead of the solid fill one.

supermerill, 2023-12-28:
I've emrged the code, and fix the arachne offsets.
next nightly should be good.

vovodroid, 2023-12-29:
> hey use the perimeter nozzle diameter instead of the solid fill one.

Actually for Arachne it shoild be extrusion width, no nozzle? Shouldn't it?

supermerill, 2023-12-29:
The extrusion width can be a % of the nozzle.

closed 2024-02-11 by supermerill