supermerill/SuperSlicer #4023: Height range modifier does not work

SirRenix, 2023-12-28:
### What happened?

When customizing skirt mods, I noticed that the infill option does not work. The percent and pattern setting does not apply to the current christmas edition.  And the pattern is shifting. Pictures attached.

### Project file & How to reproduce

Old Version
[image]
Aktuelle Christmas Version
[image]

### Version

2.5.59.4

### Operating system

Windows 11

### Printer model

Voron 2.4 Stealthburner CANBus

supermerill, 2023-12-28:
yes, that's why I added the disclaimer 'known issue : modifiers don't work' to the release

supermerill, 2023-12-28:
same source as #4016