supermerill/SuperSlicer #4025: changing 'Connection Length' infill causes crash

veehexx, 2023-12-29:
### What happened?

SuSi crashes with: `Internal error: boost::bad_any_cast: failed conversion using boost::any_cast` when changing the Connection Length away from the default 0mm setting.

[image]

### Project file & How to reproduce

1. open SuSi
2. goto Print Settings > Infill. Change 'Connection Length' to 1mm.
3. SuSi crashes with error.

### Version

2.5.59.4-rc-2ad8bf0 GTK3 Appimage

veehexx, 2023-12-29:
Same occurs in the latest current RC (commit feb5782)

supermerill, 2023-12-29:
oh, yes, i can see why.

closed 2024-02-14 by supermerill