supermerill/SuperSlicer #4026: [FR] G-code dense packing

vkhodygo, 2023-12-30:
**Is your feature request related to a problem? Please describe.**
With the default perimeter generator I get tons of consecutive code blocks like this:
```
G1 X12.345 Y12.345 E12.345
G1 F12.345
```
I also get a lot of redundant `Z` commands emitted for some reason for internal perimeters.

**Describe the solution you'd like**
To my best knowledge the standard allows it to be formatted as one line:

```
G1 X12.345 Y12.345 E12.345 F12.345
```
There are some potential [caveats](https://marlinfw.org/docs/gcode/G000-G001.html) though.

The latter should be easier to fix.
**Describe how it would work**
It's either a post-processing regex, or done on the fly.

Following the same logic, we can consolidate certain pieces of G-code for particular geometries. Say, for straight lines just keep the feedrate constant as long as possible and have one line of code instead of 10 potentially at the cost of increased print time. For lines that are parallel to *x* or *y* you don't have to send the corresponding coordinate multiple times either.

supermerill, 2023-12-31:
currently, there is a post-process that packs the gcode.
but it's currently working on the plain gcode, so it has to parse and infer. it's not very efficient.
i'll have to rework the way the gcode is lay down to improve over it.
For your exemple, it's wrong if I'm not mistaken.
```
G1 X12.345 Y12.345 E12.345
G1 F12.345
; some other commands maybe in-between but no G1/G0/G2/G3
G1 X11 Y13 E13
```
may be written as
```
G1 X12.345 Y12.345 E12.345
; some other commands maybe in-between but no move G1/G0/G2/G3 nor any custom gcode macro command
G1 X11 Y13 E13 F12.345
```

vkhodygo, 2023-12-31:
That's not my area of expertise, it could be wrong.  Still, I see a lot of repeated patterns in the preview that can be compressed.

supermerill, 2023-12-31:
I agree that it can be improved.
I won't fix it right now. It's easy to erase something that shouldn't.

vkhodygo, 2023-12-31:
No worries, that was just a comment for some time in the future. The last thing I want is to introduce a new bug.