supermerill/SuperSlicer #4027: [BUG] Arachne generates bizarre non-planar code

vkhodygo, 2023-12-30:
### What happened?

Nothing to add really, `Z` varies like there is no tomorrow, I also get twice the number of layers or something, of invalid height. 

### Project file & How to reproduce

[SuperSlicer_config_bundle.txt](https://github.com/supermerill/SuperSlicer/files/13798273/SuperSlicer_config_bundle.txt)

Try slicing a simple cube at 300% scale, check the perimeters after that.

### Version

5baf04a

### Operating system

Arch

### Printer model

ender3 v2

supermerill, 2023-12-31:
right, there is something

supermerill, 2023-12-31:
oh, just have "external_perimeter_vase" activated. This will extrude the external perimeter like a vase.

that's why you have Z on the external one.

Tell me if you find a problem with it.

vkhodygo, 2023-12-31:
Damn, that's right. Two more questions here:

 - why does it work with Arachne only though?
 - I still get much longer time than with the classic engine, any way to improve this?

supermerill, 2023-12-31:
. It was made when there was only the classic algo, so it should work with it.
. I don't know, i didn't do any extensive comparison between them. I guess Arachne can be slower because of more travels?

vkhodygo, 2024-01-01:
> It was made when there was only the classic algo, so it should work with it.

It doesn't seem so, I tried these settings on a regular cube again, for external perimeters I get only 4 lines of code in the vase mode with constant Z.

supermerill, 2024-01-16:
Can you give me your project file where the "external_perimeter_vase" don't work? i just tested on my side and nothing wrong.

vkhodygo, 2024-01-21:
It's the same config as above, I just add a regular cube and slice it.

supermerill, 2024-01-22:
can't reproduce
[image]

supermerill, 2024-01-22:
That's because you have looping perimeter activated.
You can't have "external_perimeter_vase" and it at the same time, but arachne can't have "looping perimeters", so that's why.

vkhodygo, 2024-01-22:
Fair, I wasn't aware of that. Do you thing it should toggle automatically to avoid impossible combinations of parameters? At the moment all elements simply become inactive, without any indication that certain features don't work.

supermerill, 2024-01-23:
I don't know. When you activate "looping periemters", you only have 1 perimeter, so it's printed first, whatever the setting.
I can deactivate the vase one, though.