supermerill/SuperSlicer #4028: can't see all preferences

banffman, 2023-12-30:
### Version
_Version of SuperSlicer used goes here (help->about)_
2.5.59.5

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10 Home version 22H2

### Behavior
When I go to preferences, it doesn't let me scroll through all the options in the general tab and GUI tab.  I can see there should be more preferences to adjust, because in the general tab it shows the top half of the word **paths**, and GUI has the the word **Splash Screen** but nothing below.  I tried to resize the preference window to see if it would let me scroll then, but the preferences window won't resize.  I had the same problem in SuperSlicer version 2.4.58.5 but wanted to update to the latest version before creating an issue just in case it got fixed

supermerill, 2023-12-31:
can i have a screenshot?

supermerill, 2023-12-31:
Seems like the min size can bug a bit when you change screen with a different dpi / scaling ratio.
I'll make it resizeable, until I have more information on what's happening.

banffman, 2023-12-31:
[image]
[image]

is there a different scaling ratio i should try?  i just have it set to my laptop's default screen size which is 1366x768, and SuperSlicer is maximized

supermerill, 2023-12-31:
mmm.
That's what I feared, it's not compatible anymore with 768 px columns.
I'll try to add a scroll bar.

banffman, 2023-12-31:
I appreciate that!

closed 2024-02-14 by supermerill